tohojo/sqm-scripts #131: Does not restore config for individual bridge ports at boot

patrakov, 2021-01-21:
@rulet has previously reported this as https://github.com/openwrt/luci/issues/4744 while the issue is not luci-specific at all. So, re-reporting here.

The issue is that a (strange IMHO) SQM config is applied correctly through LuCI, but not at boot. The `sqm` initscript is enabled correctly:

```
# ls -l /etc/rc.d/*sqm*
lrwxrwxrwx    1 root     root            13 Jan 21 13:05 /etc/rc.d/S50sqm -> ../init.d/sqm
```

However, there is a strange network config (a consequence of watching a possibly-misleading tutorial at https://www.youtube.com/watch?v=7982Z44ebOc):

```
config interface 'lan'
        option type 'bridge'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option ifname 'eth0.1 eth0.3'

...

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option vid '1'
        option ports '0t 2 5'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '0t 1'
        option vid '2'

config switch_vlan
        option device 'switch0'
        option vlan '3'
        option ports '0t 3 4'
        option vid '3'
```

I.e. the bridging between eth0.1 and eth0.3 is done in software, so that SQM can only be applied to one part of the LAN.

The SQM config is:

```
config queue
        option debug_logging '0'
        option verbosity '5'
        option enabled '1'
        option interface 'eth0.1'
        option qdisc_advanced '0'
        option download '70000'
        option upload '70000'
        option linklayer 'ethernet'
        option overhead '44'
        option qdisc 'cake'
        option script 'piece_of_cake.qos'

config queue
        option debug_logging '0'
        option verbosity '5'
        option enabled '1'
        option interface 'eth0.3'
        option download '0'
        option upload '0'
        option qdisc_advanced '0'
        option linklayer 'ethernet'
        option overhead '44'
        option qdisc 'cake'
        option script 'piece_of_cake.qos'
```

We have modified /etc/hotplug.d/iface/11-sqm as follows:

```
#!/bin/sh
exec >>/tmp/hotplug.log 2>&1
date
env
set -x

[ -n "$DEVICE" ] || exit 0

restart_sqm() {
    /usr/lib/sqm/run.sh stop ${DEVICE}
    /usr/lib/sqm/run.sh start ${DEVICE}
}

[ "$ACTION" = ifup ] && /etc/init.d/sqm enabled && restart_sqm

[ "$ACTION" = ifdown ] && /usr/lib/sqm/run.sh stop ${DEVICE}
```

Here is what has been logged after a reboot:

```
Thu Jan 21 13:20:11 UTC 2021
USER=root
ACTION=ifup
SHLVL=1
HOME=/
HOTPLUG_TYPE=iface
LOGNAME=root
DEVICENAME=
TERM=linux
PATH=/usr/sbin:/usr/bin:/sbin:/bin
INTERFACE=lan
PWD=/
DEVICE=br-lan
+ '[' -n br-lan ]
+ '[' ifup '=' ifup ]
+ /etc/init.d/sqm enabled
+ restart_sqm
+ /usr/lib/sqm/run.sh stop br-lan
+ /usr/lib/sqm/run.sh start br-lan
+ '[' ifup '=' ifdown ]
Thu Jan 21 13:20:12 UTC 2021
USER=root
ACTION=ifup
SHLVL=1
HOME=/
HOTPLUG_TYPE=iface
LOGNAME=root
DEVICENAME=
TERM=linux
PATH=/usr/sbin:/usr/bin:/sbin:/bin
INTERFACE=loopback
PWD=/
DEVICE=lo
+ '[' -n lo ]
+ '[' ifup '=' ifup ]
+ /etc/init.d/sqm enabled
+ restart_sqm
+ /usr/lib/sqm/run.sh stop lo
+ /usr/lib/sqm/run.sh start lo
+ '[' ifup '=' ifdown ]
Thu Jan 21 13:20:14 UTC 2021
USER=root
ACTION=ifup
SHLVL=1
HOME=/
HOTPLUG_TYPE=iface
LOGNAME=root
DEVICENAME=
TERM=linux
PATH=/usr/sbin:/usr/bin:/sbin:/bin
INTERFACE=wan
PWD=/
DEVICE=eth0.2
+ '[' -n eth0.2 ]
+ '[' ifup '=' ifup ]
+ /etc/init.d/sqm enabled
+ restart_sqm
+ /usr/lib/sqm/run.sh stop eth0.2
+ /usr/lib/sqm/run.sh start eth0.2
+ '[' ifup '=' ifdown ]
```

See - the hotplug event is about `br-lan`, while the config is about `eth0.1` and `eth0.3`. OTOH I am not sure if the use case (limiting the speed of a bridge port) is valid at all.

moeller0, 2021-01-21:
> @rulet has previously reported this as openwrt/luci#4744 while the issue is not luci-specific at all. So, re-reporting here.

Thanks for the report and time!

Mmmh, intriguing,

so https://openwrt.org/docs/guide-user/base-system/hotplug indicates:

ACTION 	“ifup”, “ifdown”, “ifupdate”
INTERFACE 	Name of the logical interface which went up or down (e.g. “wan” or “ppp0”)
DEVICE 	Physical device name which interface went up or down (e.g. “eth0.1” or “br-lan”)

so br-lan (counter to my intuition) is an expected to show up as DEVICE... I fear we might have to look at each device whether it is a bridge and whether SQM cares about any of the members?

Best Regards

tohojo, 2021-01-21:
Yeah, maybe something like:

``` diff
diff --git a/platform/openwrt/sqm-hotplug b/platform/openwrt/sqm-hotplug
index 0d56f513f208..a853b33ac59f 100755
--- a/platform/openwrt/sqm-hotplug
+++ b/platform/openwrt/sqm-hotplug
@@ -1,10 +1,18 @@
 [ -n "$DEVICE" ] || exit 0
 
+ALL_DEVICES=$(echo $DEVICE $(uci -q get network.$INTERFACE.ifname) | tr ' ' '\n' | sort -u)
+
 restart_sqm() {
-    /usr/lib/sqm/run.sh stop ${DEVICE}
-    /usr/lib/sqm/run.sh start ${DEVICE}
+    for dev in $ALL_DEVICES; do
+        /usr/lib/sqm/run.sh stop $dev
+        /usr/lib/sqm/run.sh start $dev
+    done
 }
 
 [ "$ACTION" = ifup ] && /etc/init.d/sqm enabled && restart_sqm
 
-[ "$ACTION" = ifdown ] && /usr/lib/sqm/run.sh stop ${DEVICE}
+if [ "$ACTION" = ifdown ]; then
+    for dev in $ALL_DEVICES; do
+        /usr/lib/sqm/run.sh stop $dev
+    done
+fi
```

Not sure if that will result in duplicate restarts in some cases, though?

patrakov, 2021-01-21:
I don't care about duplicate restarts. Anyway, I have talked to @rulet and convinced him that he doesn't need to use this complicated config with per-bridge-port speed limit, so he no longer has this issue. Anyway I have a spare TP-Link Archer C7 v2 and can test there if needed.

tohojo, 2021-01-22:
"Alexander E. Patrakov" <notifications@github.com> writes:

> I don't care about duplicate restarts. Anyway, I have talked to @rulet
> and convinced him that he doesn't need to use this complicated config
> with per-bridge-port speed limit, so he no longer has this issue.
> Anyway I have a spare TP-Link Archer C7 v2 and can test there if
> needed.

Right, if you do have a chance to test it that would be great. I pushed
the change to master, but will hold off on pushing it to the openwrt
packages until it's had a bit of testing :)