tohojo/sqm-scripts #132: iptables lock contention

SpareSimian, 2021-02-25:
I'm testing a new bloat test website and noticed I was getting a terrible bloat grade. When I looked at sqm-start log I discovered that a lot of the iptables commands had failed from failing to get an xlock. I stopped and started the systemd unit and re-ran the test and got an A+ bloat grade. The log showed no lock contention. I suspect that when the system boots the start script is competing with firewalld and fail2ban for the lock. The log has lines suggesting the use of the -w switch to wait for the lock.

I'm using the sqm-scripts-1.4.0-1.el7.noarch package from the EPEL repo.

Perhaps add an option to use the -w switch?

tohojo, 2021-02-25:
SpareSimian <notifications@github.com> writes:

> I'm testing a new bloat test website and noticed I was getting a
> terrible bloat grade. When I looked at sqm-start log I discovered that
> a lot of the iptables commands had failed from failing to get [...]

Which .qos script are you using?

> Perhaps add an option to use the -w switch?

Any reason not to just always use the switch?

SpareSimian, 2021-02-25:
I'm using simple.qos.

I thought perhaps omitting the switch was an attempt to avoid a deadlock. It's not an issue with a systemd unit since those run in parallel.

commit 195d3b59 (2021-02-25): functions.sh: Pass -w argument to iptables

When starting concurrently with other users of iptables, the iptables
invocations can fail because it can't grab the shared xtables lock. Pass
the -w argument to iptables to wait instead of failing, which should
hopefully make things work better. Use a 1-second max wait time to avoid
hanging forever if something goes wrong.

Fixes #132.
closed 2021-02-25 by tohojo