tohojo/sqm-scripts #133: Some modules not loaded (CentOS 7)

SpareSimian, 2021-02-26:
In my start-sqm.log I see three modules not getting loaded. They're available and I can modprobe them from the command line but not using insmod. The relevant log lines:

insmod: ERROR: could not load module act_ipt: No such file or directory
insmod: ERROR: could not load module cls_flow: No such file or directory
insmod: ERROR: could not load module sch_hfsc: No such file or directory

I get the same error using "insmod act_ipt". modprobe successfully installs that module. (I haven't tested the other two.)

uname -a shows:
Linux saruman.mpa.lan 3.10.0-1160.15.2.el7.x86_64 #1 SMP Wed Feb 3 15:06:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

sqm-scripts is sqm-scripts-1.4.0-1.el7.noarch from EPEL repo, patched to add the -w flag to iptables.

SpareSimian, 2021-02-26:
I did read through issues #3 and #13 and I'm puzzled why insmod is getting selected, since modprobe is available.

tohojo, 2021-02-26:
SpareSimian <notifications@github.com> writes:

> I did read through issues #3 and #13 and I'm puzzled why insmod is
> getting selected, since modprobe is available.

Well, defaults.sh does this:

[ -z "$INSMOD" ] && INSMOD=$(which modprobe) || INSMOD=$(which insmod)

so I suppose 'which modprobe' is failing? Maybe this is a PATH issue?

You can also override the setting in /etc/sqm/sqm.conf - just setting
the INSMOD variable should do the trick...

SpareSimian, 2021-02-26:
I may overrride it. But I'm still wondering why it's not finding modprobe. Both insmod and modprobe are in /usr/sbin and are symlinks to /usr/bin/kmod.

I did throw a 'sqm_log "INSMOD=${INSMOD}"' in do_modules while debugging this last night and indeed saw it using /usr/sbin/insmod. I'll test the which output tonight when the system isn't in use.

SpareSimian, 2021-02-27:
I see it now. This line always executes the last assignment no matter what. It's not acting like a fallback assignment as intended.

[ -z "$INSMOD" ] && INSMOD=$(which modprobe) || INSMOD=$(which insmod)

tohojo, 2021-02-27:
What shell (and version) is it being run with?

SpareSimian, 2021-02-27:
bash-4.2.46-34.el7.x86_64

I can run the line interactively and see it fail, choosing the fallback option. So you should be able to see this with any recent desktop distro. Or spin up a VPS at Linode to see it. 

I was looking at how this line is supposed to be interpreted and I think it's supposed to use the result of the which command as the status of the assignment, so it seems correct. So I'm puzzled why it's failing.

https://unix.stackexchange.com/questions/270828/how-is-the-return-status-of-a-variable-assignment-determined

gordonmessmer, 2021-02-28:
[ -z "$INSMOD" ] && INSMOD=$(which modprobe) || INSMOD=$(which insmod)

This should be re-written as:

    [ -z "$INSMOD" ] && { INSMOD=$(which modprobe) || INSMOD=$(which insmod) ; }

As it is currently written, if INSMOD is set previously, it will be replaced with the output of "which insmod".  That is, the final statement will be evaluated if either INSMOD was set previously, or if "which modprobe" exits with a non-zero status.

It's unlikely that this is a PATH issue since both modprobe and insmod are in the same directory.  The most likely explanation is that $INSMOD had a value before that line was evaluated.

tohojo, 2021-02-28:
Gordon Messmer <notifications@github.com> writes:

> [ -z "$INSMOD" ] && INSMOD=$(which modprobe) || INSMOD=$(which insmod)
>
> This should be re-written as:
>
>     [ -z "$INSMOD" ] && { INSMOD=$(which modprobe) || INSMOD=$(which insmod) ; }
>
> As it is [...]

Ah, bingo! 'start-sqm' includes defaults.sh, and simple.qos does as
well. So indeed it is evaluated twice. Nicely spotted! :)

commit 5d7e7977 (2021-02-28): defaults.sh: Don't override INSMOD variable if it is already set

@SpareSimian noticed that sqm-scripts would always use 'insmod' to load
modules on CentOS even though modprobe was also available. @gordonmessmer
noticed that this was due to the way the shell chains '&&' and '||'
operators, which would evaluate the second assignment if the first
zero-length check fails. And since defaults.sh is included from both
'start-sqm' and the .qos scripts, this double-evaluation always happens on
non-OpenWrt installations.

Fix this by properly grouping the assignment operators so they are
evaluated toge [... cut]
closed 2021-02-28 by tohojo