tohojo/sqm-scripts #134: relicensing the sqm-scripts to apache or BSD

dtaht, 2021-03-02:
I think a significant barrier to entry for some folk to "just use" the sqm-scripts has been the GPL license.

It would take the signoff of all the developers to change the license. 

I personally don't care - I like the GPL for a lot of things, but would rather people just use the sqm scripts than roll their own thing...

Opinions?

tohojo, 2021-03-02:
Dave Täht <notifications@github.com> writes:

> I think a significant barrier to entry for some folk to "just use" the
> sqm-scripts has been the GPL license.

You think or you know? These are shell scripts, distributing the source
is the default...

> It would take the signoff of all the developers to change the license. 
>
> I personally don't care - I like the GPL for a lot of things, but
> would rather people just use the sqm scripts than roll their own
> thing...
>
> Opinions?

I'd prefer to keep the GPL

dtaht, 2021-03-02:
ok. Yes, I grok they are scripts. I have run into incredible resistance in certain places about, even scripts, due to the viral nature of it. 

But gpl they can stay.

closed 2021-03-02 by dtaht