tohojo/sqm-scripts #140: Unlimited bandwidth option.

yutayu, 2021-09-12:
My Vdsl connections offers speed with wide range.
 want to set [no bandwidth shaping option].

tohojo, 2021-09-12:
yutayu ***@***.***> writes:

> My Vdsl connections offers speed with wide range.
>  want to set [no bandwidth shaping option].

If you set either upload or download speed to '0' that will turn off SQM
in that direction...

yutayu, 2021-09-12:
[pi@pi-pc sqm-scripts]$ tc -s qdisc show dev eth0
qdisc mq 0: root 
 Sent 65388 bytes 558 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
qdisc cake 0: parent :5 bandwidth unlimited diffserv3 triple-isolate nonat nowash no-ack-filter split-gso rtt 100ms raw overhead 0 
 Sent 16199 bytes 116 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
 memory used: 4364b of 14680Kb
 capacity estimate: 0bit
 min/max network layer size:           54 /    1468
 min/max overhead-adjusted size:       54 /    1468
 average network hdr offset:            5

                   Bulk  Best Effort        Voice
  thresh           0bit         0bit         0bit
  target            5ms          5ms          5ms
  interval        100ms        100ms        100ms
  pk_delay          0us         19us         16us
  av_delay          0us          1us          1us
  sp_delay          0us          1us          1us
  backlog            0b           0b           0b
  pkts                0           38           78
  bytes               0        11069         5130
  way_inds            0            0            0
  way_miss            0            7            7
  way_cols            0            0            0
  drops               0            0            0
  marks               0            0            0
  ack_drop            0            0            0
  sp_flows            0            2            1
  bk_flows            0            0            0
  un_flows            0            0            0
  max_len             0         1936           78
  quantum          1514         1514         1514

qdisc cake 0: parent :4 bandwidth unlimited diffserv3 triple-isolate nonat nowash no-ack-filter split-gso rtt 100ms raw overhead 0 
 Sent 23164 bytes 232 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
 memory used: 2304b of 14680Kb
 capacity estimate: 0bit
 min/max network layer size:           54 /    1466
 min/max overhead-adjusted size:       54 /    1466
 average network hdr offset:            8

                   Bulk  Best Effort        Voice
  thresh           0bit         0bit         0bit
  target            5ms          5ms          5ms
  interval        100ms        100ms        100ms
  pk_delay          0us         20us         15us
  av_delay          0us          3us          1us
  sp_delay          0us          2us          1us
  backlog            0b           0b           0b
  pkts                0          202           30
  bytes               0        21188         1976
  way_inds            0            0            0
  way_miss            0            8            6
  way_cols            0            0            0
  drops               0            0            0
  marks               0            0            0
  ack_drop            0            0            0
  sp_flows            0            2            0
  bk_flows            0            0            0
  un_flows            0            0            0
  max_len             0         1466           74
  quantum          1514         1514         1514

qdisc cake 0: parent :3 bandwidth unlimited diffserv3 triple-isolate nonat nowash no-ack-filter split-gso rtt 100ms raw overhead 0 
 Sent 2717 bytes 33 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
 memory used: 1326b of 14680Kb
[... 69 lines cut ...]
  target            5ms          5ms          5ms
  interval        100ms        100ms        100ms
  pk_delay          0us         62us         32us
  av_delay          0us          2us          3us
  sp_delay          0us          2us          2us
  backlog            0b           0b           0b
  pkts                0           43           67
  bytes               0         7239         4446
  way_inds            0            0            0
  way_miss            0            7            9
  way_cols            0            0            0
  drops               0            0            0
  marks               0            0            0
  ack_drop            0            0            0
  sp_flows            0            1            1
  bk_flows            0            0            0
  un_flows            0            0            0
  max_len             0          583           90
  quantum          1514         1514         1514

[pi@pi-pc sqm-scripts]$ 

Thank you for your reply.
This is stat.
I think there are some strange line.

yutayu, 2021-09-13:
[pi@pi-pc sqm-scripts]$ sudo sqm reload
Starting SQM on all configured interfaces.
Stopping SQM on eth0
Starting SQM script: layer_cake.qos on eth0, in: 0 Kbps, out: 0 Kbps
ERROR: #---- CUT HERE ----#
ERROR: get_ifb_associated_with_if failed to extrect the ifb name from:
ERROR: filter protocol all pref 10 u32 chain 0 filter protocol all pref 10 u32 chain 0 fh 800: ht divisor 1 filter protocol all pref 10 u32 chain 0 fh 800::800 order 2048 key ht 800 bkt 0 flowid 1:1 not_in_hw match 00000000/00000000 at 0 action order 1: mirred (Egress Redirect to device *) stolen index 2 ref 1 bind 1
ERROR: Please report this as an issue at https://github.com/tohojo/sqm-scripts
ERROR: Please copy and paste everything below the cut-here line into your issue report, thanks.
layer_cake.qos was started on eth0 successfully
Stopping SQM on wlan0
Starting SQM script: layer_cake.qos on wlan0, in: 51200 Kbps, out: 51200 Kbps
layer_cake.qos was started on wlan0 successfully
-----
When I set bandwidth 0 .

yutayu, 2021-09-13:
sudo sqm reload
Starting SQM on all configured interfaces.
Stopping SQM on eth0
ERROR: cmd_wrapper: tc: FAILURE (2): /usr/bin/tc qdisc del dev eth0 ingress
ERROR: cmd_wrapper: tc: LAST ERROR: Error: Invalid handle.
ERROR: cmd_wrapper: tc: FAILURE (2): /usr/bin/tc qdisc del dev eth0 root
ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
ERROR: cmd_wrapper: tc: FAILURE (2): /usr/bin/tc qdisc del dev ifb4eth0 root
ERROR: cmd_wrapper: tc: LAST ERROR: Error: Cannot delete qdisc with handle of zero.
Starting SQM script: layer_cake.qos on eth0, in: 30720 Kbps, out: 30720 Kbps
layer_cake.qos was started on eth0 successfully
Stopping SQM on wlan0
Starting SQM script: layer_cake.qos on wlan0, in: 51200 Kbps, out: 51200 Kbps
layer_cake.qos was started on wlan0 successfully
-----
And When I set bandwidth after bandwidth 0.

tohojo, 2021-09-13:
Hmm, yeah, if you set '0' in both directions I can see how that would
maybe break. That's not really a supported mode of operation (it
corresponds to just turning off sqm-scripts entirely, so why not just do
that?)

How are you connected to your VDSL connection? Is it a separate modem
connected to your router via ethernet?

yutayu, 2021-09-14:
This VDSL gets IP through dhcp via ethernet. 
I will try to stop sqm.

yutayu, 2021-09-14:
I set bandsidth 100Mbit lol.
I like sqm :P

closed 2021-09-14 by yutayu