tohojo/sqm-scripts #141: ERROR: cmd_wrapper: ip:

mrlamud, 2021-09-15:
Hello tohojo,

I'd installed sqm-scripts-git from Archlinux's AUR,

Would you direct me to correct these errors,

```
Sep 16 06:21:39 archbox systemd[1]: Created slice Slice /system/sqm.
░░ Subject: A start job for unit sqm@enp34s0.service has begun execution
░░ A start job for unit sqm@enp34s0.service has begun execution.
Sep 16 06:21:47 archbox start-sqm[979]: Starting SQM script: piece_of_cake.qos on enp34s0, in: 0 Kbps, out: 0 Kbps
Sep 16 06:21:47 archbox start-sqm[979]: ERROR: cmd_wrapper: ip: FAILURE (2): /usr/bin/ip link add name SQM_IFB_e653b type ifb
Sep 16 06:21:47 archbox start-sqm[979]: ERROR: cmd_wrapper: ip: LAST ERROR: Error: Unknown device type.
Sep 16 06:21:47 archbox start-sqm[979]: ERROR: cmd_wrapper: ip: FAILURE (2): /usr/bin/ip link add name SQM_IFB_5f318 type ifb
Sep 16 06:21:47 archbox start-sqm[979]: ERROR: cmd_wrapper: ip: LAST ERROR: Error: Unknown device type.
░░ Subject: A start job for unit sqm@enp34s0.service has finished successfully
░░ A start job for unit sqm@enp34s0.service has finished successfully.
Sep 16 06:21:47 archbox audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=sqm@enp34s0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Sep 16 06:21:47 archbox kernel: audit: type=1130 audit(1631748107.492:119): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=sqm@enp34s0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'

```

Thank you,
K. Petsiri

tohojo, 2021-09-16:
Korkiat Petsiri ***@***.***> writes:

> Hello tohojo,
>
> I'd installed sqm-scripts-git from Archlinux's AUR,
>
> Would you direct me to correct these errors,
>
>
> ```
> Sep 16 06:21:39 archbox systemd[1]: Created slice Slice /system/sqm.
> ░░ Subject: A start job for unit ***@***.*** has begun execution
> ░░ A start job for unit ***@***.*** has begun execution.
> Sep 16 06:21:47 archbox start-sqm[979]: Starting SQM script: piece_of_cake.qos on enp34s0, in: 0 Kbps, out: 0 Kbps
> Sep 16 06:21:47 archbox start-sqm[979]: ERROR: cmd_wrapper: ip: FAILURE (2): /usr/bin/ip link add name SQM_IFB_e653b type ifb
> Sep 16 06:21:47 archbox start-sqm[979]: ERROR: cmd_wrapper: ip: LAST ERROR: Error: Unknown device type.
> Sep 16 06:21:47 archbox start-sqm[979]: ERROR: cmd_wrapper: ip: FAILURE (2): /usr/bin/ip link add name SQM_IFB_5f318 type ifb
> Sep 16 06:21:47 archbox start-sqm[979]: ERROR: cmd_wrapper: ip: LAST ERROR: Error: Unknown device type.
> ░░ Subject: A start job for unit ***@***.*** has finished successfully
> ░░ A start job for unit ***@***.*** has finished successfully.
> Sep 16 06:21:47 archbox audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 ***@***.*** comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
> Sep 16 06:21:47 archbox kernel: audit: type=1130 audit(1631748107.492:119): pid=1 uid=0 auid=4294967295 ses=4294967295 ***@***.*** comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
>
> ```

Hmm, this indicates you don't have the 'ifb' module available. Are you
running the standard Arch kernel? That should be included in that
case...

...but since you're running Arch: did you happen to update the kernel
without rebooting afterwards? That breaks module autoload until you do
reboot (which is something I run into occasionally myself)...

mrlamud, 2021-09-16:
@@@Facepalm@@@@@

Thanks for pointing me out. Indeed, I'm on custom kernel.
After reboot to lts kernel, it's fine.

I'll re-comply the kernel with ifb module right now.

closed 2021-09-16 by mrlamud