tohojo/sqm-scripts #142: PLease allow setting different qdisc in upload and download

KA2107, 2021-12-05:
Hello Devs, 

THank you for the SQM scripts. I have a Belkin RT3200 (clone of Linksys E8450) router running OpenWRT SNAPSHOT. I have a Comcast Xfinity DOCSIS plan with 600 Mbps Download (usually 20% overprovisioned, to 720 Mbps) and 20 Mbps Upload (usually 20% overprovisioned, to 24 Mbps).

The Belkin RT3200 contains a Mediatek MT7622BV CPU (1.35 GHz, Dual Core, ARM v8 Cortex-A53). However when I enabled CAKE + layer_cake.qos SQM with 600000 Kbps Ingress and 20000 Kbps Egress on the WAN interface, I get only max roughly 420 Mbps Download. I figured this was because CAKE SQM does not utilize multiple CPU cores and thus restricted to the max performance of a single 1.35 GHz core of the Mediatek MT7622BV CPU.

@dtaht in OpenWRT Forum [1] suggested that I can use fq_codel on the Ingress (Download) side and CAKE on the Egress (Upload) side. However when I setup 2 separate instances of SQM in /etc/config/sqm [2], only the Egress side CAKE SQM is setup but Ingress side fq_codel does not work. Looks like the SQM Scripts do not allow setting separate QDISC on the Download and Upload side for the same interface.

Would it be possible to add such a feature to the SQM Scripts? Thank you.

[1] https://forum.openwrt.org/t/cerowrt-ii-would-anyone-care/110554/91

[2] /etc/config/sqm
`
config queue
        option interface 'wan'
        option verbosity '5'
        option qdisc_advanced '1'
        option qdisc_really_really_advanced '1'
        option egress_ecn 'NOECN'
        option squash_ingress '1'
        option squash_dscp '1'
        option debug_logging '1'
        option ingress_ecn 'ECN'
        option linklayer 'ethernet'
        option overhead '44'
        option upload '20000'
        option qdisc 'cake'
        option script 'layer_cake.qos'
        option iqdisc_opts 'ingress besteffort nat dual-dsthost'
        option eqdisc_opts 'egress diffserv4 nat dual-srchost ack-filter'
        option enabled '1'
        option download '0'

config queue
        option enabled '1'
        option interface 'wan'
        option download '600000'
        option upload '0'
        option debug_logging '0'
        option verbosity '5'
        option qdisc 'fq_codel'
        option script 'simple.qos'
        option linklayer 'ethernet'
        option overhead '44'
`

moeller0, 2021-12-05:
The easiest path forward would be for you to take simple.qos and layer_cake.qos and convert these into a hybrid taking the ingress section from simple.qos and graft this into layer_cake.qos and save this under /usr/lib/sqm/simple_layer_cake.qos... that should get you up and running (upi need to make sure to hardcode the qdisc to be fq_codel in the ingress section). That or get a beefier primary router that can sustain layer_cake.qos at the desired rate.

KA2107, 2021-12-05:
@moeller0 Thank you for the suggestion. I copied the relevant functions and created a hybrid script as you suggested. It seems to be working.

[/usr/lib/sqm/fq_codel_plus_layer_cake.qos](https://github.com/tohojo/sqm-scripts/files/7656650/fq_codel_plus_layer_cake.qos.txt)

[/etc/config/sqm](https://github.com/tohojo/sqm-scripts/files/7656657/sqm.txt)

["tc -s qdisc show" output](https://github.com/tohojo/sqm-scripts/files/7656659/tc_qdisc_show.txt)

["ip link show" output](https://github.com/tohojo/sqm-scripts/files/7656663/ip_link_show.txt)

neilsan1366, 2021-12-09:
hi i has test and he work great thanks for shared 

my config is 

/etc/config/sqm

`config queue
	option interface 'wan'
	option verbosity '5'
	option qdisc_advanced '1'
	option qdisc_really_really_advanced '1'
	option egress_ecn 'NOECN'
	option squash_ingress '1'
	option squash_dscp '1'
	option debug_logging '1'
	option ingress_ecn 'ECN'
	option linklayer 'ethernet'
	option overhead '44'
	option download '56000'
	option upload '16000'
	option qdisc 'fq_codel'
	option script 'fq_codel_plus_layer_cake.qos'
	option iqdisc_opts ''
	option eqdisc_opts 'egress diffserv4'
	option enabled '1'