tohojo/sqm-scripts #143: Restart when underlying interface restarts

jimdigriz, 2022-01-10:
[For PPP users our interface is deleted and recreated](https://github.com/systemd/systemd/issues/481#issuecomment-1007374063) so I found it helpful to add to `platform/linux/sqm@.service` under `[Install]`:

    WantedBy=sys-devices-virtual-net-%i.device

I think this will work for Ethernet devices too, though they will not come and go so there should be no impact.

There may be a better way to do this?

jimdigriz, 2022-01-10:
...of course still trying to figure out how to have it re-run when I send a SIGTERM to pppd; this only works for `systemd restart pppd-...`

tohojo, 2022-01-10:
Hmm, why doesn't it work with SIGTERM? Isn't the device still removed,
and comes back when pppd restarts? That should still trigger the virtual
unit in systemd (I think)...

jimdigriz, 2022-01-10:
...yep, I was expecting it to do so too. I'll keep fiddling and when I find something that works reliably I'll feed it back here.

tohojo, 2022-01-10:
OK, sounds good. Feel free to just open a PR to update the unit file
when you do :)

closed 2022-01-11 by tohojo