tohojo/sqm-scripts #145: sqm add wlan0 settings automatically without config.

yutayu, 2022-01-13:
I just only use sqm for eth0 , But sqm modifies wlan0 qdisc without a config file like below.

uname -a
Linux raspberrypi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux
 

journalctl | grep sqm
Aug 07 12:58:22 raspberrypi systemd[1]: Created slice system-sqm.slice.
Jan 13 15:08:13 raspberrypi start-sqm[517]: Starting SQM script: piece_of_cake.qos on wlan0, in: 85000 Kbps, out: 1000 Kbps
Jan 13 15:08:13 raspberrypi start-sqm[516]: Starting SQM script: layer_cake.qos on eth0, in: 30720 Kbps, out: 20480 Kbps
Jan 13 15:08:15 raspberrypi start-sqm[517]: piece_of_cake.qos was started on wlan0 successfully
Jan 13 15:08:15 raspberrypi start-sqm[516]: layer_cake.qos was started on eth0 successfully

yutayu, 2022-01-13:
which sqm
/usr/bin/sqm

yutayu, 2022-01-13:
In addition , I don't run sqm at startup manually.

tohojo, 2022-01-13:
What's the contents of your sqm config file?

yutayu, 2022-01-13:
# Uplink and Downlink values are in kbps
#UPLINK=1000
#DOWNLINK=20480
DOWNLINK=30720
#DOWNLINK=100000
#DOWNLINK=0
#UPLINK=810 #88
#3UPLINK=799 #87
#UPLINK=809
#DOWNLINK=20480
#DOWNLINK=25000

#UPLINK=826 #90%
#UPLINK=780
#UPLINK=734
#UPLINK=643 #70%
#UPLINK=512
#UPLINK=100000

#UPLINK=849
#UPLINK=30720
#UPLINK=0
UPLINK=20480
#UPLINK=15000
#UPLINK=10480
#UPLINK=25000
#DOWNLINK=5120
#DOWNLINK=9800
#DOWNLINK=10176 #90%
#DOWNLINK=9045 #80%
#DOWNLINK=9836 #87%
#DOWNLINK=7914 #70%
#DOWNLINK=7897 #53*149
#DOWNLINK=9805
#DOWNLINK=10600
#DOWNLINK=10240
#DOWNLINK=8480 #75%
#DOWNLINK=11400

# SQM recipe to use. For more information, see /usr/lib/sqm/*.help

#SCRIPT=simple.qos
#
#SCRIPT=simplest_tbf.qos

#SCRIPT=simplest.qos

# Optional/advanced config
SCRIPT=layer_cake.qos
#SCRIPT=piece_of_cake.qos

#ENABLED=1
#QDISC=fq_codel
#QDISC=sfq
##QDISC=fq
#QDISC=cake

#LLAM=tc_stab
#LINKLAYER=none
#LINKLAYER=ethernet
#LINKLAYER=atm
#OVERHEAD=22
#22-14
#OVERHEAD=8
#OVERHEAD=-4
#OVERHEAD=10
#STAB_MTU=2047
#STAB_MTU=1478

#STAB_TSIZE=512
#STAB_TSIZE=256
#STAB_MPU=0
#STAB_MPU=64
#ILIMIT=
#ELIMIT=
#ILIMIT=1000
#ELIMIT=1000
#ITARGET=
#ETARGET=

# ECN ingress resp. egress. Values are ECN or NOECN.
#IECN=ECN
#EECN=ECN
#EECN=NOECN
#IECN=NOECN
# Extra qdisc options ingress resp. egress
#IQDISC_OPTS=""
#EQDISC_OPTS=""
#IQDISC_OPTS="quantum 300"
#IQDISC_OPTS="perturb 10 quantum 1643"
#IQDISC_OPTS="divisor 65536 headdrop"

#EQDISC_OPTS="noecn quantum 300"
#EQDISC_OPTS="perturb 10 quantum 1643"
#EQDISC_OPTS="divisor 65536 headdrop"
#EQDISC_OPTS="mpu 64"
#IQDISC_OPTS="mpu 64"

#EQDISC_OPTS="ack-filter dual-srchost"
IQDISC_OPTS="bridged-ptm dual-dsthost nat mpu 64 nowash"
EQDISC_OPTS="bridged-ptm dual-srchost nat mpu 64"
#IQDISC_OPTS="ack-filter dual-dsthost"
# CoDel target
#TARGET=5ms

#ZERO_DSCP_INGRESS=1
#IGNORE_DSCP_INGRESS=1

ls /etc/sqm
default.conf  eth0.iface.conf  sqm.conf

tohojo, 2022-01-13:
And what's the output of each of

systemctl status ***@***.***
systemctl status ***@***.***

?

yutayu, 2022-01-14:
● sqm@eth0.service - SQM scripts for iface eth0
     Loaded: loaded (/lib/systemd/system/sqm@.service; enabled; vendor preset: enabled)
     Active: active (exited) since Fri 2022-01-14 03:10:01 JST; 13h ago
    Process: 518 ExecStart=/usr/lib/sqm/start-sqm (code=exited, status=0/SUCCESS)
   Main PID: 518 (code=exited, status=0/SUCCESS)
        CPU: 604ms

Jan 14 03:09:58 raspberrypi systemd[1]: Starting SQM scripts for iface eth0...
Jan 14 03:09:59 raspberrypi start-sqm[518]: Starting SQM script: layer_cake.qos on eth0, in: 30720 Kbps, out: 20480 Kbps
Jan 14 03:10:01 raspberrypi start-sqm[518]: layer_cake.qos was started on eth0 successfully
Jan 14 03:10:01 raspberrypi systemd[1]: Finished SQM scripts for iface eth0.

● sqm@wlan0.service - SQM scripts for iface wlan0
     Loaded: loaded (/lib/systemd/system/sqm@.service; enabled; vendor preset: enabled)
     Active: active (exited) since Fri 2022-01-14 03:10:01 JST; 13h ago
    Process: 519 ExecStart=/usr/lib/sqm/start-sqm (code=exited, status=0/SUCCESS)
   Main PID: 519 (code=exited, status=0/SUCCESS)
        CPU: 609ms

Jan 14 03:09:58 raspberrypi systemd[1]: Starting SQM scripts for iface wlan0...
Jan 14 03:09:59 raspberrypi start-sqm[519]: Starting SQM script: piece_of_cake.qos on wlan0, in: 85000 Kbps, out: 1000 Kbps
Jan 14 03:10:01 raspberrypi start-sqm[519]: piece_of_cake.qos was started on wlan0 successfully
Jan 14 03:10:01 raspberrypi systemd[1]: Finished SQM scripts for iface wlan0.

ls /etc/sqm
default.conf  eth0.iface.conf  sqm.conf

I deleted wlan0.iface.conf,before sqm runs.

yutayu, 2022-01-14:
and , I run sqm startup, It works as expected.

tohojo, 2022-01-14:
Right, you need to disable the ***@***.*** service:

systemctl disable ***@***.***

yutayu, 2022-01-14:
I did , Thanks. and solved.
but I wish automatically sqm doesn't work with wlan0.

closed 2022-01-14 by yutayu

yutayu, 2022-01-14:
lol
I remember that I enable wlan0@sqm by my own.
Excuses me.