tohojo/sqm-scripts #151: SQM and VPN speed limited (bypass possible?)

falacu, 2022-09-29:
I have an openwrt 22.03 in a RP4 with smq enabled. Another  rp4 with a wireguard vpn is connected to the openwrt. 
When i connect to the vpn i have half limited the speed compared to sqm disabled (I have 100 Mbps down and up, and with sqm I can reach 50 up and down). I couldn't find where is the problem. I want to know if is it possible to bypass in any way the sqm. I think there is no problem with the CPU,  the load doest reach 15% with Speedtest and VPN connected. Thanks! And sorry for my bad English.

tohojo, 2022-09-29:
If you're bypassing SQM for the VPN you won't get any benefit of it, so
you might as well turn it off.

As for why it affects throughput, that's hard to tell without more
details of your setup and which tests you're running to see this...

falacu, 2022-09-29:
Thanks for your quickly answer. I leave a diagram of my network and configuration of SQM and Openwrt. I used speedtest.net to verify the connection.
 VPN Speed:
80/85 mbps up and down - SQM OFF
50 mbpsup and down - SQM ON

[image]
[image]
[image]
[image]
[image]
[image]

tohojo, 2022-09-29:
And which devices are connected through the VPN? How are you running speedtest through the VPN?

falacu, 2022-09-30:
When you  ask me for more details I understood it.
I was using the vpn from the same network, in that case the 2 peers are
downloading at the same time and that is why the sqm limits just in the
middle. I think that's the point. I am trying the vpn with sqm for the
first time and I had not realized this detail.
Connecting the client from the internet should not be a problem, I did the
test and it worked at 89 Mbps.
I apologize for the silly question and the waste of time.
Thank you very much

tohojo, 2022-09-30:
Great! You're welcome! :)

closed 2022-09-30 by tohojo