tohojo/sqm-scripts #152: Error in Ubuntu arm

yutayu, 2022-09-29:
/etc/sqm$ sudo sqm start
Starting SQM on all configured interfaces.
Starting SQM script: layer_cake.qos on eth0, in: 30720 Kbps, out: 20980 Kbps
ERROR: cmd_wrapper: ip: FAILURE (2): /usr/sbin/ip link add name SQM_IFB_f013b type ifb
ERROR: cmd_wrapper: ip: LAST ERROR: Error: Unknown device type.
ERROR: cmd_wrapper: ip: FAILURE (2): /usr/sbin/ip link add name SQM_IFB_37b76 type ifb
ERROR: cmd_wrapper: ip: LAST ERROR: Error: Unknown device type.
Starting SQM script: layer_cake.qos on wlan0, in: 60000 Kbps, out: 60000 Kbps
ERROR: cmd_wrapper: ip: FAILURE (2): /usr/sbin/ip link add name SQM_IFB_3af7f type ifb
ERROR: cmd_wrapper: ip: LAST ERROR: Error: Unknown device type.
ERROR: cmd_wrapper: ip: FAILURE (2): /usr/sbin/ip link add name SQM_IFB_87c1b type ifb
ERROR: cmd_wrapper: ip: LAST ERROR: Error: Unknown device type.

 uname -a
Linux ippan-desktop 5.15.0-1015-raspi #17-Ubuntu SMP PREEMPT Mon Sep 12 13:14:51 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux

dtaht, 2022-09-29:
modprobe ifb

if that fails ifb kernel mod isn't being built by ubuntu... you'd have to build it, or install the right module...

yutayu, 2022-09-29:
Thank you for your reply.

As you say .

sudo modprobe ifb
modprobe: FATAL: Module ifb not found in directory /lib/modules/5.15.0-1015-raspi

I cannot use zram of same reason , too :(

closed 2022-09-29 by yutayu

yutayu, 2022-09-30:
After install this program , sqm works in Ubuntu arm.
sudo apt install linux-modules-extra-raspi

This is memo.