tohojo/sqm-scripts #153: Standard x86 linux install service will not start on boot

sadkisson, 2022-10-20:
I am using this script on a x86 linux install with systemd. I cannot get the script to auto start on boot. 

sudo systemctl enable sqm@wan
Created symlink /etc/systemd/system/sys-devices-virtual-net-wan.device.wants/sqm@wan.service → /usr/lib/systemd/system/sqm@.service.
Unit /usr/lib/systemd/system/sqm@.service is added as a dependency to a non-existent unit sys-devices-virtual-net-wan.device.

When I check the status after rebooting:
sudo systemctl status sqm@wan
○ sqm@wan.service - SQM scripts for iface wan
     Loaded: loaded (/usr/lib/systemd/system/sqm@.service; enabled; preset: disabled)
     Active: inactive (dead)

When I start manually after booting the script works:
sudo systemctl start sqm@wan
 sqm@wan.service - SQM scripts for iface wan
     Loaded: loaded (/usr/lib/systemd/system/sqm@.service; enabled; preset: disabled)
     Active: active (exited) since Thu 2022-10-20 21:14:19 UTC; 5s ago
    Process: 542 ExecStart=/usr/lib/sqm/start-sqm (code=exited, status=0/SUCCESS)
   Main PID: 542 (code=exited, status=0/SUCCESS)
        CPU: 370ms

Oct 20 21:14:19 router systemd[1]: Starting SQM scripts for iface wan...
Oct 20 21:14:19 router start-sqm[542]: Starting SQM script: piece_of_cake.qos on wan, in: 50000 Kbps, out: 9500 Kbps
Oct 20 21:14:19 router start-sqm[542]: piece_of_cake.qos was started on wan successfully
Oct 20 21:14:19 router systemd[1]: Finished SQM scripts for iface wan.

Thanks for making this, it works great!

tohojo, 2022-10-20:
Erm, is your network interface actually called 'wan'? That's usually an
openwrt thing...

sadkisson, 2022-10-20:
yes, i renamed the interface to wan using systemd link file in /etc/systemd/network

[Match]
MACAddress=xx:xx:xx:xx:xx:xx

[Link]
Description=Ethernet WAN Interface
Name=wan

tohojo, 2022-10-20:
Hmm, what's the output of:

systemctl | grep wan

on your system?

sadkisson, 2022-10-20:
sudo systemctl | grep wan
  sys-devices-pci0000:00-0000:00:1c.0-0000:01:00.0-net-wan.device                           loaded active plugged   82583V Gigabit Network Connection
  sys-devices-virtual-net-ifb4wan.device                                                    loaded active plugged   /sys/devices/virtual/net/ifb4wan
  sys-subsystem-net-devices-ifb4wan.device                                                  loaded active plugged   /sys/subsystem/net/devices/ifb4wan
  sys-subsystem-net-devices-wan.device                                                      loaded active plugged   82583V Gigabit Network Connection
  sqm@wan.service                                                                           loaded active exited    SQM scripts for iface wan

tohojo, 2022-10-20:
Right, could you try editing the unit file to replace this line:

WantedBy=sys-devices-virtual-net-%i.device

with

WantedBy=sys-subsystem-net-devices-%i.device

(disable the unit first, then edit it, then do 'systemctl
daemon-reload', then re-enable)

and see if that helps?

sadkisson, 2022-10-20:
The post above is with the script active after manually start. This is what it looks like on reboot before starting the script.

sudo systemctl | grep wan
  sys-devices-pci0000:00-0000:00:1c.0-0000:01:00.0-net-wan.device                           loaded active plugged   82583V Gigabit Network Connection
  sys-subsystem-net-devices-wan.device                                                      loaded active plugged   82583V Gigabit Network Connection

sadkisson, 2022-10-20:
I made the change and ran

sudo systemctl enable sqm@wan
Created symlink /etc/systemd/system/sys-subsystem-net-devices-wan.device.wants/sqm@wan.service → /usr/lib/systemd/system/sqm@.service.
Unit /usr/lib/systemd/system/sqm@.service is added as a dependency to a non-existent unit sys-subsystem-net-devices-wan.device.

I rebooted and ran systemctl status sqm@wan:

[sadkisson@router ~]$ sudo systemctl status sqm@wan

● sqm@wan.service - SQM scripts for iface wan
     Loaded: loaded (/usr/lib/systemd/system/sqm@.service; enabled; preset: disabled)
     Active: active (exited) since Thu 2022-10-20 21:55:41 UTC; 48s ago
    Process: 333 ExecStart=/usr/lib/sqm/start-sqm (code=exited, status=0/SUCCESS)
   Main PID: 333 (code=exited, status=0/SUCCESS)
        CPU: 393ms

Oct 20 21:55:40 router systemd[1]: Starting SQM scripts for iface wan...
Oct 20 21:55:40 router start-sqm[333]: Starting SQM script: piece_of_cake.qos on wan, in: 50000 Kbps, out: 9500 Kbps
Oct 20 21:55:41 router start-sqm[333]: piece_of_cake.qos was started on wan successfully
Oct 20 21:55:41 router systemd[1]: Finished SQM scripts for iface wan.

That appears to have fixed it! Thank you!

commit d22bfa87 (2022-10-20): platform/linux: Use right unit in WantedBy

Turns out the sys-devices-virtual-net unit is only created for virtual
network devices. Use the sys-subsystem-net-devices unit instead as that is
created for any type of net device.

Fixes #153.
closed 2022-10-20 by tohojo