tohojo/sqm-scripts #154: matchall support

dtaht, 2023-02-12:
I do not know how long this facility has existed, but man tc-matchall showed

EXAMPLES
       To create ingress mirroring from port eth1 to port eth2:

              tc qdisc  add dev eth1 handle ffff: ingress
              tc filter add dev eth1 parent ffff:           \
                      matchall skip_sw                      \
                      action mirred egress mirror           \
                      dev eth2

tohojo, 2023-02-12:
IIRC, the reason we're not using it is that there is (was?) OpenWrt versions where it does not exist...

dtaht, 2023-02-12:
Is it faster? Does the offload help any?

tohojo, 2023-02-12:
Theoretically, but I doubt it's much. And no, only if we offload cake as well :)

On 12 February 2023 15:41:25 CET, "Dave Täht" ***@***.***> wrote:
>Is it faster? Does the offload help any?
>