tohojo/sqm-scripts #155: After stopping the SQM speed limit, wrong information is recorded in the system log!

openwrtdiy, 2023-02-15:
After stopping the SQM speed limit, wrong information is recorded in the system log!

`Wed Feb 15 13:48:09 2023 daemon.info pppd[21282]: Terminating on signal 15
Wed Feb 15 13:48:09 2023 daemon.info pppd[21282]: Connect time 1020.9 minutes.
Wed Feb 15 13:48:09 2023 daemon.info pppd[21282]: Sent 436542716 bytes, received 213963842 bytes.
Wed Feb 15 13:48:09 2023 daemon.notice pppd[21282]: Connection terminated.
Wed Feb 15 13:48:09 2023 user.notice SQM: Stopping SQM on ppp1398
**Wed Feb 15 13:48:09 2023 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (1): /sbin/tc qdisc del dev ppp1398 ingress
Wed Feb 15 13:48:09 2023 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Cannot find device "ppp1398"
Wed Feb 15 13:48:09 2023 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (1): /sbin/tc qdisc del dev ppp1398 root
Wed Feb 15 13:48:09 2023 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Cannot find device "ppp1398"**
Wed Feb 15 13:48:09 2023 daemon.info pppd[21282]: Exit.
Wed Feb 15 13:48:09 2023 daemon.info pppoe-server[4177]: Session 1399 closed for client 80:76:93:b4:36:c2 (10.67.5.119) on br-lan
Wed Feb 15 13:48:09 2023 daemon.info pppoe-server[4177]: Sent PADT
Wed Feb 15 13:48:09 2023 daemon.warn pppoe-server[4177]: PADT for session 1399 received from 80:76:93:B4:36:C2; should be from 00:00:00:00:00:00`

openwrtdiy, 2023-02-15:
I am now using the OpenWrt-21.02 development version to compile the X86 firmware, integrating the rp-pppoe server and SQM! Using SQM for speed limit can normally manage the bandwidth of the PPP network interface!
When I stop the speed limit of the SQM network interface, the above three errors will be prompted!

`uci del sqm.$PEERNAME.interface
uci commit sqm

/usr/lib/sqm/run.sh stop $IFNAME`

moeller0, 2023-02-15:
These errors seem justified, you first delete the interface sqm is instantiated on and then stop sqm, which IMHO rightfully complains that it could not undo the ifb set-up as the interface has gone away already. The conceptually cleanest way to avoid this would be to first stop sqm, and then ppp. If you do that, do you still see errors?

moeller0, 2023-02-15:
However, I thought that after you tear down the ppp-link, OpenWrt's hotplug should already automatically stop the corresponding sqm instance via the hotplug mechanism?

openwrtdiy, 2023-02-15:
> These errors seem justified, you first delete the interface sqm is instantiated on and then stop sqm, which IMHO rightfully complains that it could not undo the ifb set-up as the interface has gone [...]

After I changed the order of the SQM command, the system log still prompts an error!
If you do not use this command /usr/lib/sqm/run.sh stop $IFNAME, there will be no prompt in the system log, but two files will remain:
/var/run/sqm/ppp735.iptables.log
/var/run/sqm/ppp735.state

openwrtdiy, 2023-02-15:
> However, I thought that after you tear down the ppp-link, OpenWrt's hotplug should already automatically stop the corresponding sqm instance via the hotplug mechanism?

If the pppoe user goes offline without calling the SQM command parameters, the SQM parameters set when ppp goes online still exist! Including TC parameters

openwrtdiy, 2023-02-15:
Let me post the code for calling the SQM command, so that it may be faster to find out the source of the problem

**pppoe-user Set SQM configuration parameters at boot time**
https://github.com/openwrtdiy/openwrt-ipk/blob/main/luci-app-pppoe-user/root/etc/init.d/pppoe-user#L29

**When the pppoe user goes online, set the SQM network interface!**
https://github.com/openwrtdiy/openwrt-ipk/blob/main/luci-app-pppoe-user/root/etc/ppp/ip-up#L77

**When the pppoe user goes offline, delete and stop the SQM network interface!**
https://github.com/openwrtdiy/openwrt-ipk/blob/main/luci-app-pppoe-user/root/etc/ppp/ip-down#L50

moeller0, 2023-02-15:
Again, I think these are valid error messages, @tohojo however since these are to be expected for some link layers that can go away arbitrarily we could consider to convert these messages into warnings instead of errors or just silence them in sqm_stop() unconditionally. Or add a silent_stop method?

tohojo, 2023-02-16:
Hmm, so the right thing to do here would be to call 'sqm stop' before
downing the interface. But since pppd doesn't seem to have a 'pre-down'
script hook I guess that's not tenable. We could add a 'cleanup' hook
that just removes leftover state files and IFB interfaces - basically
your 'silent_stop' proposal, Sebastian, but it's not really stopping
anything if the interface is gone, so better to call it 'cleanup', I
think :)

tohojo, 2023-02-16:
@openwrtdiy could you please test if #156 works for you?

openwrtdiy, 2023-02-16:
> @openwrtdiy could you please test if #156 works for you?

I'd love to test it out soon! It takes about 3~15 hours! It's night in Myanmar time

tohojo, 2023-02-16:
openwrtdiy ***@***.***> writes:

>> @openwrtdiy could you please test if #156 works for you?
>
> I'd love to test it out soon! It takes about 3~15 hours! It's night in
> Myanmar time

Sure, no rush! Note you with that PR you also have to change your ip-down script to call:

/usr/lib/sqm/run.sh cleanup $IFNAME

instead of

/usr/lib/sqm/run.sh stop $IFNAME

openwrtdiy, 2023-02-17:
> openwrtdiy ***@***.***> writes:
> Sure, no rush! Note you with that PR you also have to change your ip-down script to call: /usr/lib/sqm/run.sh cleanup $IFNAME instead of /usr/lib/sqm/run.sh stop [...]

**The system log prompt after the pppoe user disconnects by itself**

Fri Feb 17 11:01:53 2023 daemon.info pppd[28867]: Terminating on signal 15
Fri Feb 17 11:01:53 2023 daemon.info pppd[28867]: Connect time 48.8 minutes.
Fri Feb 17 11:01:53 2023 daemon.info pppd[28867]: Sent 23837430 bytes, received 10138205 bytes.
Fri Feb 17 11:01:53 2023 daemon.notice pppd[28867]: Connection terminated.
Fri Feb 17 11:01:53 2023 user.notice SQM: Stopping SQM on ppp233
Fri Feb 17 11:01:53 2023 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (1): /sbin/tc qdisc del dev ppp233 ingress
Fri Feb 17 11:01:53 2023 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Cannot find device "ppp233"
Fri Feb 17 11:01:53 2023 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (1): /sbin/tc qdisc del dev ppp233 root
Fri Feb 17 11:01:53 2023 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: Cannot find device "ppp233"
Fri Feb 17 11:01:53 2023 daemon.info pppd[28867]: Exit.
Fri Feb 17 11:01:53 2023 daemon.info pppoe-server[4108]: Session 234 closed for client 80:76:93:b4:36:c2 (10.67.0.234) on br-lan
Fri Feb 17 11:01:53 2023 daemon.info pppoe-server[4108]: Sent PADT
Fri Feb 17 11:01:53 2023 daemon.warn pppoe-server[4108]: PADT for session 234 received from 80:76:93:B4:36:C2; should be from 00:00:00:00:00:00
Fri Feb 17 11:02:08 2023 user.info mwan3track[6928]: Check (ping) failed for target "8.8.8.8" on interface man0 (eth4)
Fri Feb 17 11:02:24 2023 user.info mwan3track[6928]: Check (ping) failed for target "8.8.8.8" on interface man0 (eth4)
Fri Feb 17 11:02:54 2023 daemon.info pppoe-server[23208]: Session 1997 created for client 80:76:93:b4:36:c2 (10.67.7.205) on br-lan using Service-Name 'service-name1'
Fri Feb 17 11:02:54 2023 daemon.info pppd[23208]: Plugin rp-pppoe.so loaded.
Fri Feb 17 11:02:54 2023 daemon.info pppd[23208]: PPPoE plugin from pppd 2.4.9
Fri Feb 17 11:02:54 2023 daemon.notice pppd[23208]: pppd 2.4.9 started by root, uid 0
Fri Feb 17 11:02:54 2023 daemon.warn pppd[23208]: Connected to 80:76:93:b4:36:c2 via interface br-lan
Fri Feb 17 11:02:54 2023 daemon.info pppd[23208]: Using interface ppp1996
Fri Feb 17 11:02:54 2023 daemon.notice pppd[23208]: Connect: ppp1996 <--> br-lan
Fri Feb 17 11:02:56 2023 daemon.notice pppd[23208]: peer from calling number 80:76:93:B4:36:C2 authorized
Fri Feb 17 11:02:56 2023 daemon.notice pppd[23208]: local  IP address 10.0.0.1
Fri Feb 17 11:02:56 2023 daemon.notice pppd[23208]: remote IP address 10.67.7.205
Fri Feb 17 11:02:56 2023 user.notice SQM: Starting SQM script: simple.qos on ppp1996, in: 33000 Kbps, out: 33000 Kbps
Fri Feb 17 11:02:57 2023 user.notice SQM: simple.qos was started on ppp1996 successfully

openwrtdiy, 2023-02-17:
**After not using this command /usr/lib/sqm/run.sh cleanup $IFNAME, the system will no longer prompt errors! there will be no prompt in the system log, but two files will remain:
/var/run/sqm/ppp1996.iptables.log
/var/run/sqm/ppp1996.state**

Fri Feb 17 11:08:06 2023 daemon.info pppd[23208]: Terminating on signal 15
Fri Feb 17 11:08:06 2023 daemon.info pppd[23208]: Connect time 5.2 minutes.
Fri Feb 17 11:08:06 2023 daemon.info pppd[23208]: Sent 542118 bytes, received 2703398 bytes.
Fri Feb 17 11:08:06 2023 daemon.notice pppd[23208]: Connection terminated.
Fri Feb 17 11:08:07 2023 daemon.info pppd[23208]: Exit.
Fri Feb 17 11:08:07 2023 daemon.info pppoe-server[4108]: Session 1997 closed for client 80:76:93:b4:36:c2 (10.67.7.205) on br-lan
Fri Feb 17 11:08:07 2023 daemon.info pppoe-server[4108]: Sent PADT
Fri Feb 17 11:08:07 2023 daemon.warn pppoe-server[4108]: PADT for session 1997 received from 80:76:93:B4:36:C2; should be from 00:00:00:00:00:00
Fri Feb 17 11:09:07 2023 daemon.info pppoe-server[27892]: Session 302 created for client 80:76:93:b4:36:c2 (10.67.1.46) on br-lan using Service-Name 'service-name1'
Fri Feb 17 11:09:07 2023 daemon.info pppd[27892]: Plugin rp-pppoe.so loaded.
Fri Feb 17 11:09:07 2023 daemon.info pppd[27892]: PPPoE plugin from pppd 2.4.9
Fri Feb 17 11:09:08 2023 daemon.notice pppd[27892]: pppd 2.4.9 started by root, uid 0
Fri Feb 17 11:09:08 2023 daemon.warn pppd[27892]: Connected to 80:76:93:b4:36:c2 via interface br-lan
Fri Feb 17 11:09:08 2023 daemon.info pppd[27892]: Using interface ppp301
Fri Feb 17 11:09:08 2023 daemon.notice pppd[27892]: Connect: ppp301 <--> br-lan
Fri Feb 17 11:09:09 2023 daemon.notice pppd[27892]: peer from calling number 80:76:93:B4:36:C2 authorized
Fri Feb 17 11:09:10 2023 daemon.notice pppd[27892]: local  IP address 10.0.0.1
Fri Feb 17 11:09:10 2023 daemon.notice pppd[27892]: remote IP address 10.67.1.46
Fri Feb 17 11:09:10 2023 user.notice SQM: Starting SQM script: simple.qos on ppp301, in: 33000 Kbps, out: 33000 Kbps
Fri Feb 17 11:09:10 2023 user.notice SQM: simple.qos was started on ppp301 successfully

openwrtdiy, 2023-02-17:
Need to retest now, please wait for my reply!

tohojo, 2023-08-21:
@openwrtdiy ping on this?

openwrtdiy, 2023-08-21:
> ping on this?

I'm still online

tohojo, 2023-08-21:
Well, the PR referenced above hasn't been merged yet, so you'd have to test the contents of the PR to see any difference...

PR #156 (2023-08-22): Add cleanup action for cleaning up after a device is removed
Add a new 'cleanup' action for cleaning up SQM state after an interface that has
gone away. Refactor the stop code a bit in the process, most notably to fix
stopping in the case where no downstream shaper is configured.

Fixes #155.
closed 2023-08-22 by tohojo