tohojo/sqm-scripts #157: SQM will automatically clear the saved PPP speed limit parameters

openwrtdiy, 2023-02-24:
Whenever the PPP interface goes online, a large number of SQM parameters are executed and saved, and the speed limit is effective! But after a while, a large number of files in the /var/run/sqm/ directory will be automatically cleared! Only a small amount of pppXXX.iptables.log and pppXXX.state will be left! The configuration inside /etc/config/sqm is intact.

The system uses OpenWrt-21.02 and SQM version number v1.5.2. I have observed many times that the files in the /var/run/sqm/ directory are automatically cleared.

moeller0, 2023-02-24:
Thhis smells like hotplug triggered, anything interesting in the log? On OpenWrt something like should help:
`logread | grep hotplug`

If so the question is why hotplug trigger on down but not up events....

openwrtdiy, 2023-02-24:
> Thhis smells like hotplug triggered, anything interesting in the log? On OpenWrt something like should help: `logread | grep hotplug`
> 
> If so the question is why hotplug trigger on down but not up events....

**Interface hot-plug logs when SQM speed limit is not enabled**
2023-02-23_22:50:46 Track Interface=man5, Device=eth15, MACaddr=, IPaddr=, Hostname=KKON Action=ifup
2023-02-23_22:50:44 Track Interface=man5, Device=eth15, MACaddr=, IPaddr=, Hostname=KKON Action=connected
2023-02-23_22:50:01 Track Interface=man5, Device=eth15, MACaddr=, IPaddr=, Hostname=KKON Action=disconnected
2023-02-23_22:49:55 Track Interface=man5, Device=eth15, MACaddr=, IPaddr=, Hostname=KKON Action=ifdown
2023-02-23_22:48:57 Track Interface=man5, Device=eth15, MACaddr=, IPaddr=, Hostname=KKON Action=ifup
2023-02-23_22:48:55 Track Interface=man5, Device=eth15, MACaddr=, IPaddr=, Hostname=KKON Action=connected
2023-02-23_22:48:12 Track Interface=man5, Device=eth15, MACaddr=, IPaddr=, Hostname=KKON Action=disconnected
2023-02-23_22:48:06 Track Interface=man5, Device=eth15, MACaddr=, IPaddr=, Hostname=KKON Action=ifdown

**2023-02-24_02:30:00 After enabling SQM speed limit, immediately check that the PPP interface speed limit is successful! SQM configuration and speed limit parameters are running normally! Check the directory location /var/run/sqm/, and more than 600 files have been successfully generated !**

**Interface hotplug log that happened this morning!**
2023-02-24_10:06:58 Track Interface=man4, Device=eth14, MACaddr=, IPaddr=, Hostname=KKON Action=ifup
2023-02-24_10:06:56 Track Interface=man4, Device=eth14, MACaddr=, IPaddr=, Hostname=KKON Action=connected
2023-02-24_10:02:21 Track Interface=man3, Device=eth13, MACaddr=, IPaddr=, Hostname=KKON Action=ifup
2023-02-24_10:02:21 Track Interface=man2, Device=eth12, MACaddr=, IPaddr=, Hostname=KKON Action=ifup
2023-02-24_10:02:21 Track Interface=man3, Device=eth13, MACaddr=, IPaddr=, Hostname=KKON Action=connected
2023-02-24_10:02:20 Track Interface=man8, Device=eth18, MACaddr=, IPaddr=, Hostname=KKON Action=ifup
2023-02-24_10:02:20 Track Interface=man2, Device=eth12, MACaddr=, IPaddr=, Hostname=KKON Action=connected
2023-02-24_10:02:20 Track Interface=man8, Device=eth18, MACaddr=, IPaddr=, Hostname=KKON Action=connected

closed 2023-03-01 by openwrtdiy