tohojo/sqm-scripts #159: Additional ifb interface for low priority traffic?

transilluminate, 2023-03-21:
Hi, thanks for the awesome work - traffic shaping is working nicely on my pi4!
I’m using either the piece_of_cake.qos or the layer_cake.qos scripts.
Bufferbloat is negligible :)

However, I want to rate limit the downloads of certain internal IP addresses (Xbox / synology) with static IP addresses.
I’ve tried adding this as another ifb with a match, but doesn’t seem to work as the existing setup takes precedence.
I’m fairly new to these commands, any pointers would be appreciated to tweak the setup…

tohojo, 2023-03-21:
The choice to not support that sort of policy marking in sqm-scripts was
a deliberate one: sqm-scripts is meant to cater more to the "just turn
it on" kind of use case. You may be better off looking at something like
qosify for features like that (though I don't know if you can get that
to run on your pi4, as it's tied to openwrt):

https://git.openwrt.org/?p=project/qosify.git;a=blob;f=README

yutayu, 2023-03-21:
Hi , Robinson

You can set priority down for each packet with filter(s).
https://www.man7.org/linux/man-pages/man8/tc-cake.8.html
Though , this is not reducing bandwidth.

Yutaka.

Sent with [Proton Mail](https://proton.me/) secure email.