tohojo/sqm-scripts #161: variable UPLINK/DOWNLINK values

jimdigriz, 2023-05-01:
tl;dr version: having `systemd` read in `/etc/sqm/%I.iface.conf` as environment variables prevents `UPLINK`/`DOWNLINK` being (easily) dynamic values. One option to fix this would be to move the loading of this configuration file into the `.qos` scripts themselves. This is what I have done, but opening this ticket to highlight the problem and so the author can think if this is a problem they think should be solved in the project or something that can just live on as a documented workaround instead for others to find.

I implemented an [SNMP subagent for my xDSL modem](https://github.com/jimdigriz/mt5311) and now have visibility on the actual line sync speed:
```
$ snmpwalk -v 2c -c public 192.0.2.1 VDSL2-LINE-MIB::xdsl2ChStatusActDataRate
VDSL2-LINE-MIB::xdsl2ChStatusActDataRate.15407.xtuc = Gauge32: 6371000 bits/second
VDSL2-LINE-MIB::xdsl2ChStatusActDataRate.15407.xtur = Gauge32: 40391000 bits/second
```

The problem is `systemd` parses the configuration `/etc/sqm/%I.iface.conf` as not a script and so there is no way for me to trivially automatically use these values to shape my uplink.

My workaround has been to tweak one of the existing SQM recipe scripts like so:
```
$ diff -u /usr/lib/sqm/layer_cake.qos /usr/lib/sqm/layer_cake_dyn.qos
--- /usr/lib/sqm/layer_cake.qos	2023-04-19 09:00:46.895399539 +0000
+++ /usr/lib/sqm/layer_cake_dyn.qos	2023-04-19 09:04:30.466672470 +0000
@@ -10,7 +10,7 @@
 
 
 #sm: TODO pass in the cake diffserv keyword
-
+. "/etc/sqm/${IFACE}.iface.conf"
 . ${SQM_LIB_DIR}/defaults.sh
 QDISC=cake
```

This then means I am able to do the dynamic component in the regular user editable configuration file `/etc/sqm/%I.iface.conf`.

Here is the relevant portion of that file for me:
```
# Uplink and Downlink values are in kbps
# * We use snmpwalk as it is a pain to figure out ifIndex but we can assume a single xDSL modem.
# * If snmpwalk fails for some reason then we fall back to some (low) defaults instead.
# * At least for me, reducing the sync speed by 95% gets me close to 'actual' link speed, the further 95% is the headroom to avoid buffering
# VDSL2-LINE-MIB::xdsl2ChStatusActDataRate.13828.xtuc
UPLINK=$(snmpwalk -Ov -v 2c -c public 192.0.2.1 .1.3.6.1.2.1.10.251.1.2.2.1.2 | sed -ne '1 s/.*: // p')
UPLINK=${UPLINK:-5500000}
UPLINK=$((UPLINK/100*95))
UPLINK=$((UPLINK/100*95/1000))
# VDSL2-LINE-MIB::xdsl2ChStatusActDataRate.13828.xtur
DOWNLINK=$(snmpwalk -Ov -v 2c -c public 192.0.2.1 .1.3.6.1.2.1.10.251.1.2.2.1.2 | sed -ne '2 s/.*: // p')
DOWNLINK=${DOWNLINK:-35000000}
DOWNLINK=$((DOWNLINK/100*95))
DOWNLINK=$((DOWNLINK/100*95/1000))

# SQM recipe to use. For more information, see /usr/lib/sqm/*.help
#SCRIPT=layer_cake.qos
SCRIPT=layer_cake_dyn.qos
```

Thoughts on how to what would be best here? Documentation or change the project so this functionality is available? Can I help in any way?

moeller0, 2023-05-01:
I think you did exactly the right thing here by integrating into one of the scripts. This is not a functionality that can be generically used:
a) it requires SNMP counters (we had a similar solution for OpenWrt lantiq modem-routers, where sync speeds can be queried from the modem "blob")
b) sync speed might be uncoupled from the contracted speed, e.g. when my ISP enabled vectoring they set the sync limit to ~116/43 Mbps (which was tailored to the new nominal 100/40 higher speed plan) but they did the same as well for customers that stayed on slower speed plans like 50/10.
c) competent ISPs do not use the DSL sync to limit access speed, but implement a per-customer traffic-limiter/policer/shaper somewhere deeper in their network (DSLAMs are more like L2 switches with unusual interfaces and are prone to head of line blocking), so the derating factor (0.95 in your case) likely needs to be confirmed/adjusted for each individual ISP.

IMHO these reasons mean that users wanting to follow your example should make sure they deal with all these factors. So I think not offering such functionality out of the box is okay, as I see no way for a configuration that will work okay for the majority of users.

Question: how much do your sync speeds actually fluctuate and is seamless rate adaption enabled on your link?

Side-note: I am not sure that:
`$((UPLINK/100*95))` is ideal here, I would try `$(((UPLINK*95)/100))` the division will result in truncation, so move these after the multiplication so the truncation error will be minimized. Same for `$((UPLINK/100*95/1000))` (`$(((UPLINK*95)/(100*1000)))'? Also why not fold everything into a single invocation like:
`UPLINK=$(((UPLINK*95*95)/(100*100*1000)))`

Final note, I would use the static configuration from `sqm/%I.iface.conf` as the default value here?

jimdigriz, 2023-05-01:
> I think you did exactly the right thing here by integrating into one of the scripts. This is not a functionality that can be generically used:
> [snipped reasons]

The problem is not "can the user find 'truth'" the problem is "the user *has* (what they consider) 'truth' but are unable to use it".

Similarly you could argue for the majority of users running their own custom router firmware is too difficult and in general the tangible benefits of the effort are not worth it....ergo OpenWRT should shut up shop :)

I think there is some value is allowing `DOWNLINK`/`UPLINK` to be dynamically assigned at run time without the need for cron/systemd-timer having to rewrite the config regularly; instead it can just reload on an event (such as an SNMP trap or noticing a difference via polling).

For me, I do not mind if this is marked 'WONTFIX', fortunately it is all open source, and I was able to adapt things to meet my local needs.

> Question: how much do your sync speeds actually fluctuate and is seamless rate adaption enabled on your link?

'Yes' and 'yes'. I live in the countryside. If a squirrel farts in the vicinity, my line sync drops a few points. In the past two weeks my line has shifted its downlink speed from 49Mbps to 45Mbps, its spring, lots of squirrels about I suppose. :)

> Regarding math/truncation

Orders of magnitude, and for at least my needs, it has no effect:
```
alex@sarasti:~$ snmpwalk -v 2c -c public 192.0.2.1 VDSL2-LINE-MIB::xdsl2ChStatusActDataRate
VDSL2-LINE-MIB::xdsl2ChStatusActDataRate.15202.xtuc = Gauge32: 6371000 bits/second
VDSL2-LINE-MIB::xdsl2ChStatusActDataRate.15202.xtur = Gauge32: 40391000 bits/second

alex@sarasti:~$ bc
bc 1.07.1
Copyright 1991-1994, 1997, 1998, 2000, 2004, 2006, 2008, 2012-2017 Free Software Foundation, Inc.
This is free software with ABSOLUTELY NO WARRANTY.
For details type `warranty'. 
scale=2
6371000/100*95/100*95/1000
5749.82
6371000*95*95/(100*100*1000)
5749.82

alex@sarasti:~$ echo $((6371000/100*95/100*95/1000))
5749
alex@sarasti:~$ echo $((6371000*95*95/(100*100*1000)))
5749
```

moeller0, 2023-05-01:
> > I think you did exactly the right thing here by integrating into one of the scripts. This is not a functionality that can be generically used:
> > [snipped reasons]
> 
> The problem is not "can the user find 'truth'" the problem is "the user _has_ (what they consider) 'truth' but are unable to use it".

But you managed to "use it" by modifying the scripts, which is one of the reasons why there are shell scripts in the first place to make modifications easy.

> 
> Similarly you could argue for the majority of users running their own custom router firmware is too difficult and in general the tangible benefits of the effort are not worth it....ergo OpenWRT should shut up shop :)

I respectfully disagree. I gave a number of reasons why there is no generic solution available. And for bespoke solutions you did exactly the correct thing IMHO putting the logic into the script itself.

> 
> I think there is some value is allowing `DOWNLINK`/`UPLINK` to be dynamically assigned at run time without the need for cron/systemd-timer having to rewrite the config regularly; instead it can just reload on an event (such as an SNMP trap or noticing a difference via polling).

I, again respectfully, disagree, if one's DSL sync really changes significantly over the course of a day, one is better served by fixing this, this is not how DSL is supposed to operate (sure different sync values can change by a handful of Kbps, but that should be small enough to allow a single static rate stetting that will work for all realistic sync values).

For truly variable rate links one needs something like [cake-autorate](https://github.com/lynxthecat/cake-autorate) or [sqm-autorate](https://github.com/sqm-autorate/sqm-autorate) on top of sqm...

> 
> For me, I do not mind if this is marked 'WONTFIX', fortunately it is all open source, and I was able to adapt things to meet my local needs.

That is not my decision to make (I refer to @tohojo for this), but my argument is you made the correct choice in how to integrate that into your set-up. The alternative is to add a way to configure the method to query the rate (as we already have two proposed options query via SNMP and via the lantiq pipe) and additional de-rating factors per direction.
BTW the *-autorate scripts above both leave the configuration value in sqm's main configuration alone and simply change the rate of the running instance instead, but that only works as thee projects instantiate constant background latency measurement processes that then in turn can change shaper rates ASAP after events that indicate that the shaper rate is outdated.

> 'Yes' and 'yes'. I live in the countryside. If a squirrel farts in the vicinity, my line sync drops a few points. In the past two weeks my line has shifted its downlink speed from 49Mbps to 45Mbps, [...]

        OK, around 10% that is relatively large, I understand your motivation to not forego that potential throughput if avoidable... I just happen to think you also found the correct solution ;) (BTW, I heard reports of the effect of humidity and temperature on the sync value of lines that operate at "full-sync" (both seem inversely proportional to the sync), sitting next to my respective CO I can not really tests that, my sync is limited by the limits programmed into the DSLAM so no changes at all except for the handful of Kbps that consecutive syncs differ over simply by chance).

> Orders of magnitude, and for at least my needs, it has no effect:
>
> ```
> alex@sarasti:~$ snmpwalk -v 2c -c public 192.0.2.1 VDSL2-LINE-MIB::xdsl2ChStatusActDataRate [...]

Fair enough, this was a rather theoretical observation anyway... ;)

tohojo, 2023-05-01:
Alexander Clouter ***@***.***> writes:

> Thoughts on how to what would be best here? Documentation or change
> the project so this functionality is available? Can I help in any way?

Erm, how are you triggering restart of the systemd unit itself? And why
can't whatever process does that not simply write some new values to
/etc/sqm/${IFACE}.iface.conf first?

jimdigriz, 2023-05-01:
> Erm, how are you triggering restart of the systemd unit itself?

Remember the [fix I got you to merge](https://github.com/tohojo/sqm-scripts/pull/144)..and then the [fix you made for my fix](https://github.com/tohojo/sqm-scripts/issues/153)? That handles restarts for me when the xDSL link glitches out.

For restarting due to the line rate changing, several ways to do this, one is to create a `systemd.timer`, the other is a whole bunch of plumbing via `snmptrapd`.

> And why can't whatever process does that not simply write some new values to `/etc/sqm/${IFACE}.iface.conf` first?

As there are ten ways to skin a cat, I picked the one that involved me not having to take care writing temporary files in a way that is safe and will never cause the system to get upset.

Starting to suspect I am the only person who thinks a single shell one liner in an existing script that calls `snmpwalk` is easier that an entire separate script that uses `mktemp`, `m4`, careful use of `mv`, etc etc.

I really did think that someone stepping up and saying "hey I have a use case here for *runtime* `UPLINK`/`DOWNLINK` values" would have been interesting. If everyone thinks it is a steaming pile of hot garbage, no worries...really. Just close the ticket.

tohojo, 2023-05-01:
Alexander Clouter ***@***.***> writes:

> I really did think that someone stepping up and saying "hey I have a
> use case here for *runtime* `UPLINK`/`DOWNLINK` values" would have
> been interesting. If everyone thinks it is a steaming pile of hot
> garbage, no worries...really. Just close the ticket.

Well before we can do this we'll first need to understand your use case
properly :)

I don't think it's a bad idea to have this available, really; I just
worry that if we just make the whole config file be sourced by the
script we'll create a footgun for users who just need to set some
values.

Another option could be to make it an explicit config option you set in
the regular conf file like `USER_CONFIG=/path/to/my/script.sh`, and if
set, sqm will source that file, and you can override just the values you
want dynamically and leave the rest in the systemd environment variables?

moeller0, 2023-05-01:
Why not simply use `tc qdisc change ...` to change the rate settings of the running instances from what ever ethos polls/waits on the SNMP updates? Is it really important to have the config file represent the current value?

jimdigriz, 2023-05-01:
>> If everyone thinks it is a steaming pile of hot garbage, no worries...really. Just close the ticket. 
>
> Well before we can do this we'll first need to understand your use case properly :)

Not really sure how to more clearly state as I did above, (paraphrased) "my modem tells me how fast it is going, I want those numbers to be picked up by your project"?

> I don't think it's a bad idea to have this available, really; I just worry that if we just make the whole config file be sourced by the script we'll create a footgun for users who just need to set some values.

Probably like any project you have had your fair share of "I edited the non-config files in a way I don't understand and it broke, fix it for me".

More so as your project is a set of Shell scripts and very accessible.

This particular footgun has been in the hands of the general public for some time.

> Another option could be to make it an explicit config option you set in
> the regular conf file like `USER_CONFIG=/path/to/my/script.sh`, and if
> set, sqm will source that file, and you can override just the values you
> want dynamically and leave the rest in the systemd environment variables?

I really do not mind. I only care that there *is* a way to have these values detected at runtime.

What opened this issue with is not an expectation on the *implementation* but it describes what I am doing to workaround a limitation in my situation. Patches are afterall a form of communication, right?

jimdigriz, 2023-05-01:
> Why not simply use tc qdisc change ... to change the rate settings of the running instances from what ever ethos polls/waits on the SNMP updates? Is it really important to have the config file represent the current value?

@moeller0 a given `tc qdisc change ...` may work today, but I have no idea that it will tomorrow.

I have more faith that `UPLINK` and `DOWNLINK` will exist X years from now in this project rather than myself/everyone continuing to use `cake` as probably something else might appear.

I am also trying to maintain less code, not bring more into existence. I do not want to have to read every commit made to this project to make sure my `tc` command continues to work. This is where madness lays.

moeller0, 2023-05-01:
> @moeller0 a given `tc qdisc change ...` may work today, but I have no idea that it will tomorrow.

Why should it stop working? After all it is documented in tc's manpage:
```
      change Some entities can be modified 'in place'. Shares the
              syntax of 'add', with the exception that the handle cannot
              be changed and neither can the parent. In other words,
              change cannot move a node.
```
if this changes all bets are off and sqm-scripts will likely run into more problems with tc ;)

> I have more faith that `UPLINK` and `DOWNLINK` will exist X years from now in this project rather than myself/everyone continuing to use `cake` as probably something else might appear.

I am not sure sqm-scripts will survive a deprecation of tc ;) it mostly is a wrapper around tc and a few other commands to make it accessible for non-experts.

> I am also trying to maintain less code, not bring more into existence. I do not want to have to read every commit made to this project to make sure my `tc` command continues to work. This is where madness lays.

Now, this is clearly your prerogative, but I have a feeling you did not give this tc-change proposal an unbiased consideration, tc is unlikely to change in that regard and sqm-scripts will be in deep problems if it should change in such a way (well, we currently do not use `change` so a bug/change restricted to change would not be catastrophic). However, the tc-change method only works easily for single qdiscs per direction, for a HTB tree with multiple leaf HTB instances one would need to walk over all leave HTB instances to adjust all of them, most users however tend to use cake IIUC.

Anyway your time, your decision.

jimdigriz, 2023-05-01:
> > @moeller0 a given `tc qdisc change ...` may work today, but I have no idea that it will tomorrow.
> 
> Why should it stop working? After all it is documented in tc's manpage:
>
> [snipped]
> 
> if this changes all bets are off and sqm-scripts will likely run into more problems with tc ;)

@moeller0 You mis-understand me.

I am *not* suggesting `tc` will change how it functions in some non-backwards compatible manner.

The problem is there is no *single* `tc qdisc change ...` command that will work with all of the `*.qos` scripts. Over time schedulers may come and go...meanwhile I know I will always have a value for `UPLINK` and `DOWNLINK`.

If I went with your suggestion, then if I change the `*.qos` script, I also have to change the crafted `tc` command to work with the new script.

Meanwhile if `UPLINK`/`DOWNLINK` supported shell expansions, then the user does not need to do (or understand) anything.

moeller0, 2023-05-01:
As I said, `tc qdisc change...` will work for all solitary qdiscs... so it should work for:
1) simplest.qos
2) simplest_TBF.qos
3) piece_of_cake.qos
4) layer_cake.qos

it will not simply work with simple.qos (as there are one base and 3 leaf instances that need to be changed as well).

Yes, in the future that might change, but the alternative to the light weight tc change is really tearing down the qdisc instances and bringing them up again, considerably more disruptive than a mere rate change...

> If I went with your suggestion, then if I change the `*.qos` script, I also have to change the crafted `tc` command to work with the new script.

That depends, if it is a solitary qdisc you really only need the interface and optionally the name of the qdisc, only for hierarchical qdisc trees you would need to create your own tree walker (and figure out how to adjust the individual rates).

Again your choice what you do, but keep in mind that if changes are rare this does not matter much, but once your changes are happening often restarting an sqm instance is rather heavy handed compared to changing the shaper rate. Again, your choice.

closed 2023-05-02 by jimdigriz