tohojo/sqm-scripts #162: diffserv8 Tin's are the wrong way around or the classify is wrong

francisuk1989, 2023-05-01:
Using **diffserv8** and i found out that the Tin's are the worng way around or the classify is worng, For example

LE/CS1 will go into Best Effort
CS7 will go to Bulk

OpenWrt SNAPSHOT (r22702-cf8d861978)
Platform  ipq40xx/generic
Date 2023-05-01

dave14305, 2023-05-02:
diffserv8 uses numbered tins instead of Bulk, Besteffort, Voice, Video, etc. Can you explain how you’re using and measuring diffserv8? Or are you referring to WMM?

francisuk1989, 2023-05-02:
> Can you explain how you’re using and measuring diffserv8? Or are you referring to WMM?
Easy way to explain this is, If i use https://github.com/openwrt/qosify for example and setting;

qosify.conf

```
config defaults
	list defaults /etc/qosify/*.conf
	option dscp_icmp besteffort
	option dscp_bulk bulk

config class besteffort
	option ingress LE
	option egress LE

config class bulk
	option ingress CS7
	option egress CS7

option mode diffserv4 > diffserv8

```

00-defaults.conf

```
# WWW/SSL/SSH/DNS/NTP
tcp:80	besteffort
udp:80	besteffort
udp:443	besteffort
tcp:443	besteffort
```

Everything that isnt listed herem should go into bulk.

As you can see, LE wil give highest priority on diffserv8 then CS7, C57 i belive is mention to be high then LE

> Or are you referring to WMM?
Nothing to do with Wireless.

dave14305, 2023-05-02:
I’m still confused by your report.

- diffserv8 does not have bulk or best effort tins. It reports Tin 0 through Tin 7.
- Qosify and sqm-scripts should not be used together
- Cake diffserv8 puts LE (01) in Tin 0 and CS7 (56) in Tin 7.

Maybe someone smarter will come along, but please clarify how you observe the “bad” behavior.