tohojo/sqm-scripts #163: sqm-hotplug: 'ifname' replaced by 'device'

efahl, 2023-07-09:
In OpenWrt 22.03, `network.$INTERFACE.ifname` was removed and `ifname` was renamed to `device`, so the following code no longer functions properly:

https://github.com/tohojo/sqm-scripts/blob/master/platform/openwrt/sqm-hotplug#L3

More about this change in first warning box at https://openwrt.org/docs/guide-user/base-system/basic-networking

tohojo, 2023-07-17:
Hmm, right; that code was a bit of a hack in the first place. Maybe moving to using the 'net' hotplug (as @moeller0 suggested back when it was added) will allow us to get rid of that hack entirely? 🤔

efahl, 2023-07-17:
Yup, when I noticed the problem I started digging through the history and decided against the simple PR with a name change.  The real issue looked much deeper than I was able to understand, hence my somewhat terse "it's busted."

moeller0, 2023-07-18:
I have been keeping a low profile on this, as I am still on turrisOS which as far as I can see has not yet made that change (being based on OpenWrt21). So I can not actually test any proposed changes for new functionality, only for backward compatibility...

tohojo, 2023-08-21:
OK, to answer my own question, the 'net' hotplug event only gives us 'add' and 'remove' events, so we can't just substitute that. We'll have to query the device sections instead...

tohojo, 2023-08-21:
@moeller0 and @efahl could you please test #164 and see if that works correctly (it's totally untested as I don't have a compatible config myself to test this on right now)...

moeller0, 2023-08-21:
I will be able to test the back-ward compatibility later this week, still on turrisOS which has not made the step to OpenWrt22. 
How do I properly test this? Instantiate sqm-scripts on a LAN interface, but do I force a hotplug event?

tohojo, 2023-08-21:
`ifup`/`ifdown` should work I think :)

openwrtdiy, 2023-08-21:
> @moeller0 and @efahl could you please test #164 and see if that works correctly (it's totally untested as I don't have a compatible config myself to test this on right now)...

**I just tested SQM with OpenWrt-23.05! Now submit the SQM system log**

Mon Aug 21 14:16:39 2023 user.notice SQM: Starting SQM script: piece_of_cake.qos on wan, in: 85000 Kbps, out: 10000 Kbps
Mon Aug 21 14:16:39 2023 user.notice SQM: piece_of_cake.qos was started on wan successfully
Mon Aug 21 14:21:36 2023 daemon.warn odhcpd[1864]: No default route present, overriding ra_lifetime!
Mon Aug 21 14:21:59 2023 user.notice SQM: Stopping SQM on wan
Mon Aug 21 14:22:00 2023 user.notice SQM: Starting SQM script: piece_of_cake.qos on lan4, in: 37500 Kbps, out: 12500 Kbps
Mon Aug 21 14:22:01 2023 user.notice SQM: piece_of_cake.qos was started on lan4 successfully
Mon Aug 21 14:23:50 2023 user.notice SQM: Stopping SQM on lan4
Mon Aug 21 14:23:51 2023 user.notice SQM: Starting SQM script: piece_of_cake.qos on lan4, in: 12500 Kbps, out: 37500 Kbps
Mon Aug 21 14:23:52 2023 user.notice SQM: piece_of_cake.qos was started on lan4 successfully
Mon Aug 21 14:25:31 2023 user.notice SQM: Stopping SQM on lan4
Mon Aug 21 14:25:32 2023 user.notice SQM: Starting SQM script: piece_of_cake.qos on br-lan, in: 25000 Kbps, out: 12500 Kbps
Mon Aug 21 14:25:32 2023 user.notice SQM: piece_of_cake.qos was started on br-lan successfully
Mon Aug 21 14:26:45 2023 user.notice SQM: Stopping SQM on br-lan
Mon Aug 21 14:26:45 2023 user.notice SQM: Starting SQM script: piece_of_cake.qos on br-lan, in: 12500 Kbps, out: 25000 Kbps
Mon Aug 21 14:26:46 2023 user.notice SQM: piece_of_cake.qos was started on br-lan successfully
Mon Aug 21 14:29:20 2023 daemon.warn odhcpd[1864]: No default route present, overriding ra_lifetime!
Mon Aug 21 14:30:45 2023 user.notice SQM: Stopping SQM on br-lan
Mon Aug 21 14:30:46 2023 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0, in: 37500 Kbps, out: 12500 Kbps
Mon Aug 21 14:30:47 2023 user.notice SQM: piece_of_cake.qos was started on eth0 successfully
Mon Aug 21 14:32:22 2023 user.notice SQM: Stopping SQM on eth0
Mon Aug 21 14:32:22 2023 user.notice SQM: Starting SQM script: piece_of_cake.qos on lan4, in: 25000 Kbps, out: 37500 Kbps
Mon Aug 21 14:32:23 2023 user.notice SQM: piece_of_cake.qos was started on lan4 successfully
Mon Aug 21 14:33:22 2023 daemon.warn odhcpd[1864]: No default route present, overriding ra_lifetime!
Mon Aug 21 14:33:24 2023 kern.info kernel: [ 1280.676382] mt7530-mdio mdio-bus:1f lan4: Link is Down
Mon Aug 21 14:33:24 2023 kern.info kernel: [ 1280.683659] br-lan: port 4(lan4) entered disabled state
Mon Aug 21 14:33:24 2023 daemon.notice netifd: Network device 'lan4' link is down
Mon Aug 21 14:33:25 2023 daemon.notice netifd: bridge 'br-lan' link is down
Mon Aug 21 14:33:25 2023 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Aug 21 14:33:25 2023 daemon.warn odhcpd[1864]: No default route present, overriding ra_lifetime!
Mon Aug 21 14:35:08 2023 kern.info kernel: [ 1385.418297] mt7530-mdio mdio-bus:1f lan4: Link is Up - 1Gbps/Full - flow control off
Mon Aug 21 14:35:08 2023 kern.info kernel: [ 1385.426160] br-lan: port 4(lan4) entered blocking state
Mon Aug 21 14:35:08 2023 kern.info kernel: [ 1385.431375] br-lan: port 4(lan4) entered forwarding state
Mon Aug 21 14:35:08 2023 daemon.notice netifd: Network device 'lan4' link is up
Mon Aug 21 14:35:08 2023 daemon.notice netifd: bridge 'br-lan' link is up
Mon Aug 21 14:35:08 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Aug 21 14:35:09 2023 kern.info kernel: [ 1386.235835] mt7530-mdio mdio-bus:1f lan4: Link is Down
Mon Aug 21 14:35:09 2023 kern.info kernel: [ 1386.242751] br-lan: port 4(lan4) entered disabled state
Mon Aug 21 14:35:09 2023 daemon.notice netifd: Network device 'lan4' link is down
Mon Aug 21 14:35:09 2023 daemon.notice netifd: bridge 'br-lan' link is down
Mon Aug 21 14:35:09 2023 daemon.warn odhcpd[1864]: No default route present, overriding ra_lifetime!
Mon Aug 21 14:35:09 2023 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Aug 21 14:35:12 2023 kern.info kernel: [ 1389.195747] mt7530-mdio mdio-bus:1f lan4: Link is Up - 1Gbps/Full - flow control rx/tx
Mon Aug 21 14:35:12 2023 kern.info kernel: [ 1389.203736] br-lan: port 4(lan4) entered blocking state
Mon Aug 21 14:35:12 2023 kern.info kernel: [ 1389.208993] br-lan: port 4(lan4) entered forwarding state
Mon Aug 21 14:35:12 2023 daemon.notice netifd: Network device 'lan4' link is up
Mon Aug 21 14:35:12 2023 daemon.notice netifd: bridge 'br-lan' link is up
Mon Aug 21 14:35:12 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Aug 21 14:35:12 2023 daemon.warn odhcpd[1864]: No default route present, overriding ra_lifetime!
Mon Aug 21 14:35:13 2023 daemon.warn odhcpd[1864]: No default route present, overriding ra_lifetime!
Mon Aug 21 14:35:16 2023 daemon.warn odhcpd[1864]: No default route present, overriding ra_lifetime!
Mon Aug 21 14:35:20 2023 daemon.warn odhcpd[1864]: No default route present, overriding ra_lifetime!
Mon Aug 21 14:35:29 2023 daemon.warn odhcpd[1864]: No default route present, overriding ra_lifetime!
Mon Aug 21 14:37:15 2023 authpriv.info dropbear[6556]: Child connection from 192.168.95.239:12172
Mon Aug 21 14:37:17 2023 authpriv.notice dropbear[6556]: Auth succeeded with blank password for 'root' from 192.168.95.239:12172
[... cut ...]
Mon Aug 21 15:02:42 2023 user.notice root:  guest  SPEED-LIMIT >>>>>> The script has been executed!
Mon Aug 21 15:03:00 2023 authpriv.info dropbear[8403]: Child connection from 192.168.95.239:12407
Mon Aug 21 15:03:01 2023 authpriv.notice dropbear[8403]: Auth succeeded with blank password for 'root' from 192.168.95.239:12407
Mon Aug 21 15:03:14 2023 authpriv.info dropbear[8417]: Child connection from 192.168.95.239:12410
Mon Aug 21 15:03:14 2023 authpriv.notice dropbear[8417]: Auth succeeded with blank password for 'root' from 192.168.95.239:12410

tohojo, 2023-08-22:
Awesome, thank you for testing! I'll merge this and see if we can still get it into OpenWrt 23.05 before the final release :)

PR #164 (2023-08-22): openwrt/hotplug: Handle DSA switch port configs
In the OpenWrt hotplug script we handle the case where the interface hotplug event arrives for the bridge interface by querying uci to find the underlying devices that are part of the bridge and restarting any configured SQM instances on the ports. However, with the switch to DSA, we can no longer find the bridge ports in the network.$INTERFACE.ifname property. Instead, we have to query the 'device' property and locate the separate device config section to find the port names.

Add a function to do this querying, but keep the ifname query to reta [... cut]
closed 2023-08-22 by tohojo