tohojo/sqm-scripts #166: sqm scripts with rpi4 bookworm errors

yutayu, 2023-10-30:
[   40.016392] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   40.020674] bcmgenet fd580000.ethernet eth0: Link is Down
[   43.076921] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   43.077352] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   47.224732] bcmgenet fd580000.ethernet eth0: hw csum failure
[   47.224755] skb len=56 headroom=184 headlen=56 tailroom=1936
               mac=(130,14) net=(144,40) trans=184
               shinfo(txflags=0 nr_frags=0 gso(size=0 type=0 segs=0))
               csum(0xffff4449 ip_summed=2 complete_sw=0 valid=0 level=0)
               hash(0x0 sw=0 l4=0) proto=0x86dd pkttype=0 iif=2
[   47.224772] dev name=eth0 feat=0x0000010000004829
[   47.224779] sk family=10 type=2 proto=17
[   47.224787] skb headroom: 00000000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224794] skb headroom: 00000010: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224801] skb headroom: 00000020: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224807] skb headroom: 00000030: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224812] skb headroom: 00000040: 80 7f b0 00 80 80 00 00 23 7d 00 00 00 00 00 00
[   47.224818] skb headroom: 00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[   47.224823] skb headroom: 00000060: 92 fe 09 bd a9 85 ab 44 bc ed 2b f5 d3 23 dd 4b
[   47.224828] skb headroom: 00000070: a7 a5 1f c1 93 c2 f6 33 d8 d5 20 96 d1 f0 39 b7
[   47.224834] skb headroom: 00000080: 00 00 dc a6 32 89 fb bd 00 50 f1 80 00 00 86 dd
[   47.224839] skb headroom: 00000090: 60 00 00 00 00 38 11 34 20 01 03 a0 00 00 20 05
[   47.224845] skb headroom: 000000a0: 00 00 00 00 00 57 01 23 24 07 c8 00 5a 31 07 00
[   47.224850] skb headroom: 000000b0: 6e 59 55 f0 28 f2 c9 65
[   47.224855] skb linear:   00000000: 00 7b 8a 89 00 38 74 f4 24 02 06 e7 00 00 0c 56
[   47.224861] skb linear:   00000010: 00 00 07 41 85 f3 ec 12 e8 ea 47 d9 6d 57 c0 96
[   47.224866] skb linear:   00000020: 07 72 0c 80 26 69 94 5d e8 ea 4b 28 4c ee a3 31
[   47.224872] skb linear:   00000030: e8 ea 4b 28 4c f3 33 5f
[   47.224877] skb tailroom: 00000000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224882] skb tailroom: 00000010: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224888] skb tailroom: 00000020: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224893] skb tailroom: 00000030: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224899] skb tailroom: 00000040: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224904] skb tailroom: 00000050: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224909] skb tailroom: 00000060: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224915] skb tailroom: 00000070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224920] skb tailroom: 00000080: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224925] skb tailroom: 00000090: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224931] skb tailroom: 000000a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224936] skb tailroom: 000000b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224941] skb tailroom: 000000c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224947] skb tailroom: 000000d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224952] skb tailroom: 000000e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224958] skb tailroom: 000000f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224963] skb tailroom: 00000100: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224969] skb tailroom: 00000110: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224974] skb tailroom: 00000120: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224980] skb tailroom: 00000130: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224985] skb tailroom: 00000140: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224990] skb tailroom: 00000150: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.224996] skb tailroom: 00000160: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225001] skb tailroom: 00000170: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225006] skb tailroom: 00000180: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225012] skb tailroom: 00000190: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225017] skb tailroom: 000001a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225023] skb tailroom: 000001b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225028] skb tailroom: 000001c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225033] skb tailroom: 000001d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225038] skb tailroom: 000001e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225044] skb tailroom: 000001f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225049] skb tailroom: 00000200: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225054] skb tailroom: 00000210: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225060] skb tailroom: 00000220: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225065] skb tailroom: 00000230: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225071] skb tailroom: 00000240: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225076] skb tailroom: 00000250: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225081] skb tailroom: 00000260: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225087] skb tailroom: 00000270: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225092] skb tailroom: 00000280: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[   47.225097] skb tailroom: 00000290: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[... 96 lines cut ...]
[   47.225972]  __netif_receive_skb+0x20/0x70
[   47.225979]  netif_receive_skb+0x164/0x1d0
[   47.225987]  ifb_ri_tasklet+0x160/0x2ac [ifb]
[   47.225999]  tasklet_action_common.constprop.0+0x12c/0x15c
[   47.226008]  tasklet_action+0x40/0x50
[   47.226015]  __do_softirq+0x198/0x4d8
[   47.226020]  ____do_softirq+0x18/0x24
[   47.226025]  call_on_irq_stack+0x24/0x54
[   47.226030]  do_softirq_own_stack+0x24/0x3c
[   47.226036]  __irq_exit_rcu+0xd4/0x120
[   47.226043]  irq_exit_rcu+0x18/0x50
[   47.226050]  el1_interrupt+0x38/0x70
[   47.226056]  el1h_64_irq_handler+0x18/0x2c
[   47.226061]  el1h_64_irq+0x64/0x68
[   47.226065]  arch_cpu_idle+0x18/0x2c
[   47.226071]  default_idle_call+0x54/0x19c
[   47.226076]  do_idle+0x26c/0x2c0
[   47.226082]  cpu_startup_entry+0x30/0x3c
[   47.226086]  rest_init+0xe8/0xf0
[   47.226092]  arch_post_acpi_subsys_init+0x0/0x28
[   47.226099]  start_kernel+0x704/0x744
[   47.226104]  __primary_switched+0xbc/0xc4

 uname -a
Linux raspberrypi 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux

tohojo, 2023-10-30:
This looks like a bug in the kernel NIC driver - nothing do to with sqm...

yutayu, 2023-10-31:
I see.Ifb causes it.
Waiting for new driver.

SpareSimian, 2023-10-31:
@yutayu, Is there a bug link we can track? Maybe at Debian or RPi?

yutayu, 2023-11-01:
> @yutayu, Is there a bug link we can track? Maybe at Debian or RPi?
I tried :P