tohojo/sqm-scripts #170: change simple.qos to alow cake as leaf qdisc?

moeller0, 2024-02-17:
https://github.com/tohojo/sqm-scripts/blob/405aeffa48e6385df4bb267e2a2fac6e254f96f8/src/simple.qos#L94

Currently, if qdisc is set to cake, simple.qos will set up something close to layer_cake.qos. The intent was at the time to be able to allow simple A/B tests between HTB+fq_codel and cake+cake; that is using cake as both traffic shaper and fq_AQM. Now, today anybody wanting to do that can simply use layer_cake.qos instead. So I would like to propose to remove this special casing from simple.qos to allow HTB+cake. The rationale for that is that a number of users resort to not using cake as traffic shaper as it has issues on some platforms and it would be sweet to have a simple testing tool again to probe whether the issue is caused by cake"s shaper or by the rest of cake"s features.

tohojo, 2024-02-20:
Well, I do get the desire to allow easy A/B testing, and wouldn't mind having this work differently. However, changing the default at this point will impact everyone who is currently running simple.qos with the qdisc set to 'cake' - i.e., they will get different behaviour than they have before. Not sure if it's a good idea to change things from under people in that way?

moeller0, 2024-02-20:
Fair enough, easy to remedy with a simple_HTB.qos addition with essentially just that change (and some clean-up). My guess is nobody uses simple.qos as vehicle to use cake, people typically mention layer_cake.qos or piece_of_cake.qos, but as I said this is just a guess...