tohojo/sqm-scripts #171: On OpenWRT, initscript's `reload` is called too frequently/when not necessary

brianjmurrell, 2024-04-19:
Hi.  There's a [ticket open in OpenWRT](openwrt/packages#22996) about an observation that _sqm-scripts_ reloads frequently and when it's unnecessary.

I think I've debugged this issue here to the point of understanding that it's happening when my IPv6 WAN connection (which is not even the interface I have SQM configured for!) triggers an `ifupdate` event.  I believe that when this happens https://github.com/tohojo/sqm-scripts/blob/33a89d8e7f6c82acc0adced84cf5cd95447e6bb5/platform/openwrt/sqm-init#L6-L9 causes `procd` to issue a `/etc/init.d/sqm reload`.

But clearly this is happening on far too many unrelated events.

Is there any way to do some more checking in [`/etc/init.d/sqm`](https://github.com/tohojo/sqm-scripts/blob/33a89d8e7f6c82acc0adced84cf5cd95447e6bb5/platform/openwrt/sqm-init) to limit the reload to only happen when necessary?

tohojo, 2024-04-20:
Well, AFAICT [from the documentation](https://openwrt.org/docs/guide-developer/procd-init-scripts#specifying_triggers), that `procd_add_reload_trigger` thing should only cause a reload if `/etc/config/sqm` is changed. There's a separate set of commands to add monitoring of interfaces through procd, but we're not using those (we have the hotplug script instead). So I'm not really sure how sqm ends up being reloaded, actually...

moeller0, 2024-04-20:
My guess is that the ISP might do something unusual with how IPv6 is supplied here that tickles OpenWrt the wrong way......

tohojo, 2024-04-20:
On 20 April 2024 13:44:46 CEST, moeller0 ***@***.***> wrote:
>My guess is that the ISP might do something unusual with how IPv6 is supplied here that tickles OpenWrt the wrong way......

Maybe, but that still doesn't explain why the ifupdate event leads to a reload of sqm...

brianjmurrell, 2024-04-20:
> Well, AFAICT [from the documentation](https://openwrt.org/docs/guide-developer/procd-init-scripts#specifying_triggers), that `procd_add_reload_trigger` thing should only cause a reload if `/etc/config/sqm` is changed.

Right!?!?

> There's a separate set of commands to add monitoring of interfaces through procd, but we're not using those (we have the hotplug script instead). So I'm not really sure how sqm ends up being reloaded, actually...

Indeed.  It is puzzling.  Seems like a bug TBH.

> Maybe, but that still doesn't explain why the ifupdate event leads to a reload of sqm...

Exactly, which is why it seems like a bug.  But what is the OpenWRT component this is a bug with?  procd?

tohojo, 2024-04-21:
"Brian J. Murrell" ***@***.***> writes:

>> Maybe, but that still doesn't explain why the ifupdate event leads to a reload of sqm...
>
> Exactly, which is why it seems like a bug.  But what is the OpenWRT
> component this is a bug with?  procd?

I think procd is just relaying events; my guess would be that there is
some other component triggering this. No idea which :/