tohojo/sqm-scripts #173: `sqm-scripts-extra` - version number update needed?

richb-hanover, 2024-06-13:
I saw a note - https://github.com/openwrt/packages/issues/23706 - that details a switch of OpenWrt to using the APK package manager in place of the unmaintained `opkg`. That issue lists a number of packages whose version number doesn't match the required `major.minor.fix` format. (See the note for details.)

I see that it lists `sqm-scripts-extra`. I'm guessing that this is relevant to this repo. Does anything need to be done? Many thanks.

richb-hanover, 2024-06-13:
Adding @hnyman - It appears that `sqm-scripts-extra` hasn't been updated for 8 years. https://github.com/openwrt/packages/tree/openwrt-23.05/net/sqm-scripts-extra/src 

- Are they still relevant? 
- Does anything need to be done for a switch to APK?

Thanks again

hnyman, 2024-06-16:
The ancient package `sqm-scripts-extra` could be removed.
I added it 8 years ago as cake was under development and there was a need to provide easier access to it before it came officially here. Later cake has been officially adopted into SQM, (and the cake qdisc itself into Linux).

I will remove the package.