tohojo/sqm-scripts #175: [Request] A build for custom openwrt fw with 5ghz support for archer c7 v1 (kernel 4.19)

NoUserNameForYou, 2024-10-01:
Hello, I really really appreciate your work. Thanks to your efforts gaming is much more tolerable during rush hours! So thanks very much! I'm currently using it in openwrt 23.05 and it works wonders!

I have a request and I'd understand if it's a hassle for you to do it. See thing with archer c7 v1 is its 5ghz doesn't work out of the box. Someone made patches and made a compiled firmware which enables 5ghz and it works fine. However its kernel is old and sqm scripts or the luci-app-sqm won't work on it. 

Could you please make an sqm scripts and luci-app-sqm for this build linked here? https://github.com/d687r02j8g/openwrt_c7v1/raw/master/ath10k.zip (related discussion topic with more info https://forum.openwrt.org/t/archer-c7-v1-0-crashes-on-boot/19784/ )

I tried installing sqm scripts different older versions but none of them succeeded. When I try it from the luci web software section I'm getting kernel mismatch error. I believe the installed kernel is 4.19, does your work need a backport for that kernel?

Thanks in advance

tohojo, 2024-10-01:
We can't supply custom packages, but the dependency of the sqm-scripts
package on a particular kernel version is incidental. You should be able
to install the package from a newer openwrt version just fine using
`opkg --force-depends`. You may not be able to use the 'cake' qdisc if
your build doesn't include it, but simple.qos and simplest.qos should
work just fine with fq_codel :)

NoUserNameForYou, 2024-10-01:
Thanks for getting back. Will let you know how it goes in a day or so, I'll do it in my free time.

NoUserNameForYou, 2024-10-02:
I got this error even with the force depends but after trying once more with the same command it installed sqm scripts. 

[code]Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for sqm-scripts:
 *      kernel (= 6.6.52~29623f3f324cc7a2e4543c558766d4ef-r1)[/code]

Luci-app-sqm also gave these errors

[code] * check_data_file_clashes: Package luci-lib-base wants to install file /usr/lib/lua/luci/debug.lua
        But that file is already provided by package  * luci-base
 * check_data_file_clashes: Package luci-lib-base wants to install file /usr/lib/lua/luci/http.lua
        But that file is already provided by package  * luci-base
 * check_data_file_clashes: Package luci-lib-base wants to install file /usr/lib/lua/luci/ltn12.lua
        But that file is already provided by package  * luci-base
 * check_data_file_clashes: Package luci-lib-base wants to install file /usr/lib/lua/luci/util.lua
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package luci-app-sqm.
[/code]

through the putty console but trying once more over the luci gui made it install.
Also it complains about missing xtables-nft and xtables-legacy, I see they were migrated into the firewall app, I looked for them in older repos but couldn't find. Is it possible for you to share them?

However, sqm management page on luci shows sqm disabled and enable button won't work. :(

NoUserNameForYou, 2024-10-02:
Tc -s qdisc provides this

```
Error relocating /sbin/tc: __time64: symbol not found
Error relocating /sbin/tc: __gettimeofday_time64: symbol not found
Error relocating /sbin/tc: __localtime64: symbol not found
Error relocating /sbin/tc: __lstat_time64: symbol not found
```
and the var/run/sqm/available_qdiscs folder is empty.

tohojo, 2024-10-02:
You'll have to install the tc package from the build you're using. Only the sqm-scripts package itself can be installed across versions

On 2 October 2024 14:38:05 CEST, NoUserNameForYou ***@***.***> wrote:
> Tc -s qdisc provides this
>
>
> ```
> Error relocating /sbin/tc: __time64: symbol not found
> Error relocating /sbin/tc: __gettimeofday_time64: symbol not found
> Error relocating /sbin/tc: __localtime64: [...]

NoUserNameForYou, 2024-10-02:
Thanks so much my friend. Thanks for leading me in the correct path, I got it to work. Now all is complete!

If another person wants to learn how to do it

1. Replace the contents of etc/opkg/distfeeds with these lines

```
src/gz openwrt_kmod http://mirror-03.infra.openwrt.org/releases/19.07.9/targets/ath79/generic/kmods/4.14.267-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d
src/gz openwrt_core http://mirror-03.infra.openwrt.org/releases/19.07.9/packages/mips_24kc/packages
src/gz openwrt_base http://mirror-03.infra.openwrt.org/releases/19.07.9/packages/mips_24kc/base
src/gz openwrt_luci http://mirror-03.infra.openwrt.org/releases/19.07.9/packages/mips_24kc/luci
src/gz openwrt_packages http://mirror-03.infra.openwrt.org/releases/19.07.9/packages/mips_24kc/packages
src/gz openwrt_routing http://mirror-03.infra.openwrt.org/releases/19.07.9/packages/mips_24kc/routing
src/gz openwrt_telephony http://mirror-03.infra.openwrt.org/releases/19.07.9/packages/mips_24kc/telephony
src/gz openwrt_ex2 http://downloads.openwrt.org/releases/18.06.8/targets/ar71xx/generic/kmods/4.9.214-1-2b8f9dfe583e5c09aadd9474da55137f
src/gz openwrt_ex3 http://downloads.openwrt.org/releases/18.06.8/targets/ar71xx/generic/packages
src/gz openwrt_ex4 http://archive.openwrt.org/releases/18.06.0/targets/ar71xx/generic/packages
src/gz openwrt_ex5 http://mirror-03.infra.openwrt.org/releases/18.06.0/packages/mips_24kc/packages
src/gz openwrt_ex7 http://mirror-03.infra.openwrt.org/releases/18.06.0/packages/mips_24kc/base
src/gz openwrt_ex8 http://mirror-03.infra.openwrt.org/releases/18.06.0/packages/mips_24kc/luci
src/gz openwrt_ex9 http://mirror-03.infra.openwrt.org/releases/18.06.0/packages/mips_24kc/routing
src/gz openwrt_ex10 http://mirror-03.infra.openwrt.org/releases/18.06.0/packages/mips_24kc/telephony
src/gz openwrt_ex11 http://mirror-03.infra.openwrt.org/releases/18.06.0/targets/ar71xx/generic/packages
```
2. Get the public keys from this page https://openwrt.org/releases/18.06/start and this page https://openwrt.org/releases/19.07/start and rename them. Remove the .txt extension, make them extensionless and remove the "usign_" from the beginning of the file name. Then put them in the etc/opkg/keys folder. Winscp is a good app for that. 
3. Then refresh repos with opkg update command through putty app.
4. Then type  `opkg install --force-depends --force-overwrite luci-app-sqm` it'll whine about bunch of things but it'll install.

tohojo, 2024-10-02:
NoUserNameForYou ***@***.***> writes:

> Thanks so much my friend. Thanks for leading me in the correct path, I
> got it to work. Now all is complete!

Great! You're welcome :)

NoUserNameForYou, 2024-10-02:
I give up. It won't work and I'm too tired to try more things. Somehow the fw has 4.19 kernel and is made from a snapshot or something. Official openwrt pages list 4.14 and closest 5.4 kernel, jumping over 4.19. 

Thanks for trying to help at least!

closed 2024-10-02 by NoUserNameForYou